iNavFlight/inav #8569: FW (safety feature) - Landing heading after RTH with rc lost, or execute a WP mission

robustini, 2022-11-21:
## Current Behavior

I have not found this featureand on seems to me that there have been other requests like this.
It would be very useful that with fixed wings after an RTH due to unrecoverable problems on the tx/rx system the plane would perform autolanding in a definable heading direction.

## Desired Behavior

RTH is triggered for an unrecoverable problem at the rx, such as it stops working due to firmware or connection problems
An RTH is then performed as per the settings, and when the plane arrives on the home it performs the usual Loiter and then performs autoland by leveling the wings in a heading direction (definable in degrees), so as to avoid dangerous directions.

## Suggested Solution

Many people do not utiilize to WP navigation, so it would be even more useful if in case of RTH you could declare a mission to be executed, in that case so could schedule an automatic landing by setting a route.

## Who does this impact? Who is this for?

To all who may have an unrecoverable link loss during the flight.

## Additional context

cfitches, 2023-03-08:
I want this too. Sorry I duplicated it here: https://github.com/iNavFlight/inav/issues/8855

b14ckyy, 2023-03-09:
I looked into a similar approach and wrote down a full functional concept. Also tested with logic programming (although this is not practical for daily use) as a proof of concept. 
https://github.com/iNavFlight/inav/issues/8623

b14ckyy, 2024-03-20:
https://github.com/iNavFlight/inav/pull/9713

closed 2024-03-20 by b14ckyy