iNavFlight/inav #8571: Continous Servo trim acts wrong on V-Tail Mixes

b14ckyy, 2022-11-21:
## Current Behavior
If a Plane with a V-Tail mix has Continous Servo trim active, it is likely that, for the Pitch Axis, the servos are trimmed in opposite directions. I was finally able to reproduce and it seems like this happens, if Pitch has a negative + Positive mixed servo and no Inverted output. Both servo 2 and 3 are trimmed into the same direction (signal wise) but that means that one control surface goes up and the other one goes down. 

[image]
[image]

## Steps to Reproduce

1. Set up a V-Tail plane with same configuration: Mixer with one pitch servo inverted, outputs both V-Tail servos uninverted
2. Go fly with Yaw Integral at 0 to disable Continous trim on that axis
3. Enable continous servo trim
4. Check control surfaces after landing

## Expected behavior
Pitch axis continous trim goes into the correct direction no matter what Mixer or Servo direction is set on each Servo. 

## Suggested solution(s)
Take mixer and output reversal both into account for trimming Pitch on a V-Tail.

## Additional context
I will try to get more Information from other pilots and compare their mixer and output settings with mine to find common factors. 
Here is a Log of the last flight with Servotrim Debug enabled: 
[LOG00040.zip](https://github.com/iNavFlight/inav/files/10054890/LOG00040.zip)

---

- FC Board name and vendor: Any
- INAV version string: Issue started to appear with INAV 3.0 and Continous Servo trim feature implemented.

b14ckyy, 2022-11-22:
@Jetrell as far as I am concerned this does not really seem to help. The Continous servo trim does not touch yaw at all if the Integral on yaw is set to 0. But the pitch trim seems to trim the servos in the opposite way. At least it looks like it. 

All planes reported so far have in the outputs tab either both V-Tail servos reversed or none reversed. Never differently. I will try another test flight when possible but then reverse one servo only and compensate with the mixer to see how that behaves.

Jetrell, 2022-11-23:
Here is the mixers on a Mini Talon.. Noticed that the Pitch stabilization servo's both have positive weight, compared to your example. Which is what you thought could be an issue.. But by not having Stabilized Yaw. It makes it easier to get the Pitch mixing correct in this manor.
This Talon has no issues running Continuous servo trim.

[image]

[image]

Jetrell, 2022-11-23:
> Here is the mixers on a Mini Talon.. Noticed that the Pitch stabilization servo's both have positive weight. Which is what said is the issue. 
But by not having Stabilized Yaw. It makes it easier for people to get the Pitch mixing correct in this manor. This Talon has no issues running Continuous servo trim.
> 
> [image]
> 
> [image]

b14ckyy, 2022-11-23:
@Jetrell This is really interesting. 
you ar ethe only one so far not reporting trim issues and your V-Tail servos are running the opposite in the outputs tab (one inverted, one not)
Allt eh people reporting issues including myself are running both servos either inverted or uninverted and the direction change happens in the Mixer. 
So I really need to change that in my setup and doublecheck if the issue still persists then. If it does we should have tracked down the root cause.

Jetrell, 2022-11-23:
That's why I thought the simplification of just using RC Yaw is helping to workaround the problem. Because you don't have to take into account the Yaw stabilization response.  Only RC Yaw control.
I've experienced the same issue with a nano talon a few releases back.. But I don't have that plane and its setup, to now compare.

> So I really need to change that in my setup and double check if the issue still persists then. If it does we should have tracked down the root cause.

Let me know what you find.

b14ckyy, 2022-11-23:
So far all people reporting the same issue have, had the mixes and reverses set as in the example above. 
Either both V-Tail servos unreversed or both reversed and all correct movements regulated in the mixer. 

I have now changed that. Reversed Servo 3 (right Tail servo) so it runs opposite to the left servo 2 from the outputs and changed the mixer accordingly to have all directions correct again. Another test pilot did the same and goes for a test flight right now. 

[image]
[image]

b14ckyy, 2022-11-23:
OK it behaves the same with this configuration. So next step would be to disable yaw entirely and see what happens.

kasatka60, 2022-11-26:
[image]
[image]
[image]
inav 6.0 fp1

b14ckyy, 2022-12-03:
OK we have confirmation now. The wrong trim behavior is unrelated to yaw. Shawn made a test flight today with no yaw mix. Both servos plain pure Pitch mixed. Still the left servo, as always, is trimmed down instead of up like the right servo. 

so there is just a trim value inverted for dual servo pitch controls. @avsaase do you think you can have a look if you have overlooked something back then? You know the code best.

closed 2023-03-06 by DzikuVx