iNavFlight/inav #8572: Add Servo output configuration to programming interface

fernandodlc, 2022-11-21:
## Current Behavior
Currently you cannot use programming logic to modify trims or servo configuration.
Use Case: I have a nano drak with 2 sets of wings, I want to use the logic conditions to modify the servo setup and mixing to switch between the 2. The linkage geometry is different for each set of wings, the larger wings will be flown with a tail. I need to setup the trims and limits for the servos differently for each set of wings but I am unable to do so. I don't need to reverse any servos but I can see how this could have been the case.

## Desired Behavior
Ability to apply a logic condition to change the trims, limits and servo reversing.

## Suggested Solution
The ability to apply a logic condition like in the mixer would be ideal

## Who does this impact? Who is this for?
Anyone who wants to have 2 different wing setups on the same plane eg draks

fernandodlc, 2022-11-21:
Alternatively - This kind of feature wouldn't be out of place under the battery profile selection. Quite often batteries of different weights will require different trim settings.

b14ckyy, 2022-11-21:
this is all possible. In the mixer tab you have all the way to the right a flag to enable a mixer based on a logic condition. 
So you set up one mix set that is enabled by LCx and another mix set that is enabled by LCy
at the same time you can switch PID Profiles with logic conditions to also adapt the tune and rates witht he same Logic. 

I am planning the same with a Nano Drak that I want to fly in 3 different configs. 
28" pure, 28" boom and 40" boom. 

for the trims these should be adjusted mechanically so the software trim in inav is as close as possible to 1500 for each setup. If you fly in Acro, then use Continous servo trim to fine adjust it in flight. If you usually fly in manual, then just let the plane fly in angle or acro mode for a few seconds after launch with sticks centered before you swith to manual mode.

fernandodlc, 2022-11-21:
@b14ckyy I disagree, the mixer settings yes but not the servo output settings. 

1. On my setup for example I have the pushrods perpendicular to the control surface on the 40" and in line with the airflow on the 28". This means that mechanically the servos move the control surfaces differently on the 2 wings. I'd like to limit the servo travel on one set of wings but not the other.
2. Trims *should* be as close to 1500 as possible but this is often tricky to do in practice. Not all pushrods are easy to adjust. In the tailless setup the amount of reflex is fairly crucial for a successful launch so getting the trims right is important and I'd like to ensure the trims are distinct in both instances.
3. Lastly there is always the situation of damage or general physical issues throwing things out. Picking up a clump of grass on the winglets could throw off the auto-trim setting the trim to an unacceptable value for the other set.

b14ckyy, 2022-11-21:
@fernandodlc 

1. is not a problem at all as you can also limit the throw in the mixer with the same effect. 

2 and 3 I see your point, although that's only a problem if you really only want to fly manual exclusively. Both would just need a few seconds of stick center flight in acro or angle to trim. Not sure if this few seconds of inconvenience after a wing swap (I guess you don't swap them for every single launch) justifies the work to add multiple Output profiles. This would not only raise the firmware size but also the config size and might kill off F722 and F411 FCs even sooner. For something that only 0.01% of pilots might need. As I said I am planning the same and I would not other with waiting for the trim to be done. Especially as I fly mostly in Acro. 

Just thinking loud.

fernandodlc, 2022-11-21:
I could be wrong but limiting the throws in the mixer will limit the overall throw. In the case I describe the amount of movement in one direction is greater than the other.

b14ckyy, 2022-11-21:
Ah okay got ya. so you want to limi the throw in just one direction and for that either change the endpoints or the center position. Makes sense then. 
But then it should not be in the battery profiles but more in the PID Profiles. You need to switch these anyways between wings. 
Maybe this could be done as part of the VTOL Project? @shota3527

fernandodlc, 2022-11-21:
yes exactly. I've got both the battery and pid profiles changing between setups so that works.

I kinda like the way you can set certain parameters with "feature BAT_PROF_AUTOSWITCH" that would be a really convenient way to handle this in my case.

b14ckyy, 2022-11-21:
Yeah I use that on my Dart 250G as I fly it on 2S liion sometimes for cruising or on 4S 850mAh for ripping around. But on 4S it needs different pids as well. So it switches automatically depending on the battery.

shota3527, 2022-11-22:
Do you mean you would like to change these parameters between different wings?
```
servo 1 850 2300 2000 100
servo 2 700 2150 950 100
servo 3 1200 1800 1500 -100
servo 4 1000 2000 1494 -100
```

The `smix` and `mmix` can be switched by the underdeveloping mixer_profile feature #8555 
It is possible to add servo into the mixer_profile as well.

But since changing the wing takes some time, you can copy/paste the CLI commands after the installation.

@b14ckyy 
Do you think inav should consider something like hardware switching/replacing without opening the configurator?

b14ckyy, 2022-11-22:
@shota3527 imho it would make sense. @fernandodlc and myself are already 2 persons who are building a modular plane with quick swappable Wing/Tail Configs. 
I think having some kind of way to switch the Outputs Configuration, would already help. Maybe similar to the Mixer Switches based on logic conditions? 
This might also help with things like switchable Pan and Tilt configs. I have a plane that will have 2 pan and tilt setups I want to switch in between and this would allow me to have different servo ranges for both setups. For the nose one for example I have just a small up-range but bigger down-range with a center offset. On the dome tilt at the bottom of the plane I want it to be linear. I already wondered How I could do that.

closed 2023-10-27 by DzikuVx