iNavFlight/inav #8574: Inflight Adjustment - selected item in osd flashes too slowly (once every 8 seconds)

jreise-d, 2022-11-22:
INAV 6.0.0 FP1, Matek F722-wpx

When I use the "in flight adjustment", the selected parameter flashes on the osd screen. Since INAV 6.0.0 it flashes very slowly, once in 8 seconds. Normally it should flash at 1 Hz or so. It seems that a timer is set wrong here

b14ckyy, 2022-11-23:
Do you know how to flash the firmware for the PixelOSD chip? 

if yes flash the following firmware to your OSD. Flasher for each OS is also there. 
https://1drv.ms/u/s!Ar6QsGgoXRgquKhnL2EVC3YhClf8uw?e=FS6rEP
then set the baud rate in the Ports tab for the PixelOSD to 250000. 

this should work and make the blinking much faster.

jreise-d, 2022-11-23:
Yes, I know it. Actually my PixelOSD runs with 2.0.0. Thanks, I will try that.

b14ckyy, 2022-11-23:
FYI: This is a development firmware I got from FrSky before they cancelled the whole project and deleted the github. It was intended to minimize the artefacting issues in combination with the higehr baud rate. I hope this also solves your blinking issue.

jreise-d, 2022-11-23:
I tested this, with the 2.0.90 it's the same issue. I noticed that other alarms in osd (e.g. sat count) first are blinking correct. But when I select a parameter in the adjustment, then this parameter and also the sat count are blinking at very low period. When deselecting the parameter, sat count blinks normal again.

b14ckyy, 2022-11-23:
Ah okay that's interesting. I also noticed slow blink with in-flight adjustments although not 8s cycles. 
This is how it looks for me although this is 3.0 https://youtu.be/vdhg54bsiEM?t=461

jreise-d, 2022-11-23:
yes, that is the normal behavior i had in my previous version (2.6.1). In 6.0.0 I have about 8s on and 8s off, too slow to work with.

b14ckyy, 2022-11-23:
I will check the next time I fly. Not sure if I used the adjustments on the AR Pro on 6.0 so far.

jreise-d, 2022-11-23:
ok. And thanks for your help so far!

it would be a pity if the in-flight-adjustment would no longer run on boards with pixelosd

OptimusTi, 2022-12-27:
Similar issue here. Not a pixel board. MATEKF411.

OptimusTi, 2022-12-31:
I believe the issue starts after e7a894b2942828c5a1599f7ebf0a2f496be5762f

tonyyng, 2023-04-02:
I'm using INAV 6.0.0 with DJIWTF and I see a similar problem.  The blinking interval is 8s on and 8s off, whether or not adjustments are enabled.  

No change with display_force_sw_blink = ON

tonyyng, 2023-04-04:
After some investigation, I found a logic change that fixed the problem for me: https://github.com/iNavFlight/inav/blob/master/src/main/drivers/display.c#L69
```
#define SW_BLINK_CYCLE_MS 200 // 200ms on / 200ms off
static bool displayEmulateTextAttributes(displayPort_t *instance, char *buf, size_t length, textAttributes_t *attr) {
  TEXT_ATTRIBUTES_REMOVE_BLINK(*attr);
    if ((millis() / SW_BLINK_CYCLE_MS) % 2) {
        memset(buf, ' ', length);
        buf[length] = '\0';
        // Tell the caller to use buf
        return true;
    }
    // Tell the caller to use s but with the updated attributes
    return false;
```
I found that if I changed SW_BLINK_CYCLE_MS from 200 to 20, the OSD blinking interval went from 12s off, 12s on to a very usable 700ms off, 700ms on.

I had several OSD elements blinking, a couple with warnings and one with inflight adjustment, and they were all blinking using the same interval.

I don't understand why this works.   The original logic looked okay.  

I have display_force_sw_blink = OFF, but it looks like DJIWTF doesn't support character blinking, so it is going through the software blinking logic anyway.

This works for my particular configuration (HGLRCF7 and DJIWTF), but isn't appropriate as a general solution.  Hopefully, this gives someone more knowledgeable more insight into the problem and we can get a general fix for this issue.

tonyyng, 2023-04-05:
I checked another model of mine that is running INAV 4.1, MATEKF405TE and wtfos.  With 4.1, there isn't a specific video system setting for WTFOS.  I'm using HD (HDZERO?).  The blinking with this combination is also slow; a little over 4 seconds off and 4 seconds on.  

There was no change in code I posted above between 4.1 and 6.0.

OptimusTi, 2023-04-05:
Something between 4 and 4.1 breaks soft blinking. I played with SW_BLINK_CYCLE_MS. My slow blinking is more in the 2 second window.

tonyyng, 2023-04-05:
> Something between 4 and 4.1 breaks soft blinking. I played with SW_BLINK_CYCLE_MS. My slow blinking is more in the 2 second window.

What OSD system are you using?

OptimusTi, 2023-04-05:
Analog

glennvenghaus, 2023-09-26:
Hi, I also have this problem with WTFOS , DJI VISTA's , Goggle V2 (all latest versions of msp-osd/wtf) and inav 6.0, 6.1 and 7.0 dev build. Flywoo 405pro/nano stacks.
Is there any way to fix/bypash this issue as the extrene slow flashing is making my long range flights nerve wracking were 8 or 10 seconds periods the most important (as in alarm state) elements are not visible

glennvenghaus, 2023-09-26:
fyi I tried above change from @tonyyng on a local inav 7.0 build for targets flywoof405nano and geprc_f722_aio and fixed it for both quads. So doubt  its FC specific.

b14ckyy, 2024-03-20:
Can you please test with 7.1 RC1?

closed 2026-09-22 by b14ckyy