iNavFlight/inav #8576: 6.0 FP1 Matek F405-WSE SBUS input is broken

ABLomas, 2022-11-23:
Only noticed on one model - Matek F405-WSE - so far.
SBUS input here is broken - only few channel values are read, independent on UART (all UARTS do the same), others are "fixed".
Tested on two FC's, one just out of package, the same issue.
Other FC's tested - Airbot F4, Matek F722-SE - do not have this issue, the same RX+TX is read correctly via SBUS.

JulioCesarMatias, 2022-11-23:
duplicate of https://github.com/iNavFlight/inav/issues/8566

ABLomas, 2023-01-29:
> duplicate of #8566

No.
They had issues on 5.1 (unclear which ones), my issue on 6.0 and still not fixed with RC1
Makes SBUS unusable on F405-WSE with new release.

ABLomas, 2023-01-31:
Oh and forgot to add - with INAV 5.1 it's all fine.

closed 2024-03-22 by b14ckyy