iNavFlight/inav #8578: Any plan for supporting DJI Goggles 2 canvas mode?

donne1226, 2022-11-24:
I wanna switching from v1 to Goggles 2

FlyChenPiPi, 2022-11-28:
The canvas mode of INAV6.0 does not work properly when using DJI Goggles V2 with O3 air unit

b14ckyy, 2022-11-30:
its not canvas mode. It is MSP-Displayport. DJI names it wrong there causing lots of confusion. What did you set up in ports? If it works as expected then you should only enable MSP-Displayport in the Peripherals column on the UART. Do not enable MSP on the left column.

OlivierC-FR, 2022-11-30:
> its not canvas mode. It is MSP-Displayport. DJI names it wrong there causing lots of confusion. What did you set up in ports? If it works as expected then you should only enable MSP-Displayport in the Peripherals column on the UART. Do not enable MSP on the left column.

I tried that 5 mins ago, does not work, there's no OSD in my Goggle V2.
The O3 only display on OSD the various video bandwidth and settings, and the battery voltage it's connected to.

For clarification, here talking about brand new V2 goggles (not G2), no rooted, flashed with newest (official) firwmare so they can connect to the O3 air unit also with newest firmware.

OlivierC-FR, 2022-11-30:
I'm gonna flash Betaflight and see how it goes...

FlyChenPiPi, 2022-11-30:
> > its not canvas mode. It is MSP-Displayport. DJI names it wrong there causing lots of confusion. What did you set up in ports? If it works as expected then you should only enable MSP-Displayport in the Peripherals column on the UART. Do not enable MSP on the left column.
> 
> I tried that 5 mins ago, does not work, there's no OSD in my Goggle V2. The O3 only display on OSD the few datas for the video link (bandwidth, LQ), and the battery voltage it's connected to.
> 
> For clarification, here talking about brand new V2 goggles (not G2), no rooted, flashed with newest (official) firwmare so they can connect to the O3 air unit also with newest firmware.

Yes, I tried 5.1 and 6.0 but could not display OSD information normally. The OSD working mode of O3 seems to be only compatible with Betaflighe4.3, which caused a lot of trouble for my remote FPV UAV.

donne1226, 2022-11-30:
> its not canvas mode. It is MSP-Displayport. DJI names it wrong there causing lots of confusion. What did you set up in ports? If it works as expected then you should only enable MSP-Displayport in the Peripherals column on the UART. Do not enable MSP on the left column.

I just ordered it today, and I will try it out when I got it.

OlivierC-FR, 2022-11-30:
Obviously the topic is rather wide, and not limited to the initial title, there's many different kind of goggles (V1, V2, G2, rooted and WTFOS, not rooted, etc), there's the new O3 air units, the old AU/Vistas etc, maybe we should make a whole new topic for clarification. Also, in some cases/combos the solution is only in DJI hands, not INAV side.

brousselle, 2022-11-30:
I also followed the exact same way and tests as @OlivierC-FR but with the new Googles 2.
And i confirm there is no OSD.

b14ckyy, 2022-11-30:
Then I have no idea. This O3 stuff and Goggles 2 is brand new stuff. DJI didn't gave a sh** in the past about INAV. I wouldn't be surprised if it is the same here. 
although MSP-DP is standardized for the 3 existing systems, maybe they messed up the implementation for O3, who knows.

MrD-RC, 2022-11-30:
It's likely DJI need to implement the font. But I'd have thought you'd at least get weirdness in the goggles. Either way. I feel at this stage it's something that people should be complaining to DJI about. We know MSP DisplayPort is working. Even with an old firmware on Avatar, I got glitching OSD characters.

OlivierC-FR, 2022-11-30:
At least the MSP for arm is going through, the O3 detects the arming and react accordingly by starting and stopping the onboard recording, but the MSPs about the OSD are not ok obviously.

b14ckyy, 2022-11-30:
> At least the MSP for arm is going through, the O3 detects the arming and react accordingly by starting and stopping the onboard recording, but the MSPs about the OSD are not ok obviously

When people talking about MSP its always confusing. You are NOT supposed to use MSP if they use the same OSD type as wtf.OS, Walksnail and HDZero. 
send a screenshot of your ports tab please.

claudiozavagno, 2022-11-30:
I can confirm that the dji unit gets the arm input, starting the recording, if i set the peripheral as dji fpv vtx but also if i set the peripheral as HDzero VTX. Enclosing the port screen. No OSD on either, though.
[image]

OlivierC-FR, 2022-11-30:
> When people talking about MSP its always confusing. You are NOT supposed to use MSP if they use the same OSD type as wtf.OS, Walksnail and HDZero. send a screenshot of your ports tab please.

Can't SS right now, MSP is not enabled on the left-most column, here talking about the column Peripheral set to MSP displayport.
And as Claudio said above, setting to DJI FPV or HDzero as trigger a response of the O3 to arming-disarming action.

See this parallel topic I made, with Betaflight 4.3.2, only checking the MSP column is enough to get the OSD  : 
https://github.com/iNavFlight/inav/issues/8596

b14ckyy, 2022-11-30:
@OlivierC-FR if for betaflight enabling MSP is enough, then they do NOT use MSP Display Port. 
It looks to me like they use the MSP Canvas that was already used in the old DJI system. 

In this case Preipherals have to be set to DJI and not to MSP-Displayport/HDZero VTX and the old DJI hack has to be used.

OlivierC-FR, 2022-12-01:
> @OlivierC-FR if for betaflight enabling MSP is enough, then they do NOT use MSP Display Port. It looks to me like they use the MSP Canvas that was already used in the old DJI system.
>
> In this case [...]

Ok I'm digging this way now, I see BF has bumped their API minor to 45 instead of 42, maybe that's only that, and/or the FC identifier, will make a build later today

claudiozavagno, 2022-12-01:
Don't know if it can be of any use but setting msp on data column gets the arm reaction from the O3, exactly as setting hdzero or dji fpv on the peripherals column
[image]

MrD-RC, 2022-12-01:
That will be an MSP connection, not a DisplayPort connection. DJI has never been set up like that on INAV. For the DJI port, MSP in the column should be off. Then in the last, peripherals, column. Either HDZero (5.1 and earlier) or MSP DisplayPort (6.0 onwards) for MSP DisplayPort; or DJI OSD for basically what they did last time.

OlivierC-FR, 2022-12-01:
Ok this is what I tried, with a custom build with  MSP API minor version set to 45 same as BF 4.3.2, instead of 42

MSP on, periph = None, speed 115200, response to arm = YES, OSD = no
MSP on, periph = None, speed 232400, response to arm = YES, OSD = no
MSP on, periph = None, speed 250000, response to arm = YES, OSD = no

MSP off, periph = DJI FPV VTX, speed 115200, response to arm = YES, OSD = no
MSP off, periph = DJI FPV VTX, speed 232400, response to arm = YES, OSD = no
MSP off, periph = DJI FPV VTX, speed 250000, response to arm = YES, OSD = no

MSP off, periph = MSP displayport, speed 115200, response to arm = no, OSD = no
MSP off, periph = MSP displayport, speed 232400, response to arm = no, OSD = no
MSP off, periph = MSP displayport, speed 250000, response to arm = no, OSD = no

MSP off, periph = Frsky OSD, speed 115200, response to arm = no, OSD = no

Next I'll try to also change the FC identifier

a1axx, 2022-12-01:
Hey, won't the character set have to match bf 4.3? 

Here is what DJI's 'canvas' (MSP-DP) can display: 
[image]

b14ckyy, 2022-12-01:
@a1axx that was already done with the old system. THere is a DJI Compatible switch in the OSD setup. 
But that's the problem. The available items are still not adequate for INAV especially for fixed wing pilots. And if it is different now from the old system it means a shitload of extra work for INAV devs and again needs hacks applied in INAV to make DJI accept the OSD data. this is utter BS imho. Just shows again how little DJI cares about us.

OlivierC-FR, 2022-12-01:
I might be wrong but I don't thing it has anything to do with the charsets, with BF 4.3 it's obviously the same old "DJI FPV" kind of communication, MSP based, not displayport, same as the old MWOSD as sample, over the UART goes the low level MSP queries and requests, I don't see why the fonts should be involved, no ?
As sample the O3 request LAT GPS, BF replies "45.12345", the O3 displays it with whatever font it want, and that's all.
It's not BF sending strings involving fonts.

Again, for the OSD to work with BF 4.3.2 it only requires the MSP column checked, nothing else.
Not displayport. Just some old MSP API versioned 1.45
Unless I misunderstood something.

b14ckyy, 2022-12-01:
@OlivierC-FR 

> As sample the O3 request LAT GPS, BF replies "45.12345", the O3 displays it with whatever font it want, and that's all.
It's not BF sending strings involving fonts.

this iis true for the "classic" DJI OSD. maybe they still use parts of that. But they also advertise full "Canvas Mode" that would actually need that char set and would work similar to MSP Display Port. 

And something MUST have changed in the DJI system. even if it still works with betaflight now. INAV got the DJI OSD hack implemented that actually fakes BF 4.2 MSP for the air unit. As you proved, this does not work anymore so they must have changed something in the firmware to prevent INAV applying this hack in the MSP side instead of actively supporting INAV. 

This char set that @a1axx posted above is not even the full betaflight OSD Char set so even there it is limited. This is just laughable.

That means again the INAV devs have to spend many hours in adapting to this. possibly changing the OSD setup page again to make different options for the old and new DJI osd set etc....

a1axx, 2022-12-01:
> @OlivierC-FR
>
>
> this iis true for the "classic" DJI OSD. maybe they still use parts of that. But they also advertise full "Canvas Mode" that would actually need that char set and would work similar [...]

To be fair it's not that bad. Users should be able to end up with an AHI & access to the stick menu. The crazy thing is, no matter how measly you think it is, I am sure a fair portion of inav users will still want to use it.

b14ckyy, 2022-12-01:
Especially that's why I don't get why DJI acts like this.... 
So many users they left behind. Especially on fixed wings this O3 air unit might be of use. Right now no one with INAV can really use it at all. That's just stupid especially now that they have serious competition with Walksnail. The O3 is just a niche product in the first place as it is just overkill in price to put on any craft. It is only suitable for very specific crafts in the first place or if the user has just too much money to throw away.

OlivierC-FR, 2022-12-01:
> Especially that's why I don't get why DJI acts like this.... So many users they left behind. Especially on fixed wings this O3 air unit might be of use. Right now no one with INAV can really use it [...]

The very root of the issue is that they sent the first batch of gear to the same dozen of influencers, half of them make a video for the $ then put it on a shelf and never use it again, the other half fit on a BF quad and do like 3 flights "wooaaaaa product is awesome !!!" and that's the end of it. If only they would send one or two samples to INAV and arduplane users or devs, even the worse moron would think to send them an email like "eh what about adding support for everything than a BF quad"... same sad old story, again. Sorry for the rant, they never learn.

b14ckyy, 2022-12-01:
They have not made the old DJI systems INAV and Ardu compatible even after 3 years. Do you seriously think that this will change after they now seem to actively block INAV? 
I mean. Nothing changed in BF. It works with BF 4.2 and later. INAV has the DJI OSD hack implemented that fakes a BF 4.2 FC. but now it does not work anymore so there must be an active block in place.

MrD-RC, 2022-12-01:
@a1axx was this the fix Josh?

_I had to add O subcommand (heartbeat i think) to get it working on ultra. NO OTHER HD SYTEMS NEEDED IT.
my bad, i didnt know that, i used logic analyser to sniff tradffic between bf and 03_

a1axx, 2022-12-01:
@MrD-RC yes I understand this was the fix. 

He used a logic sniffer and found this was the issue or missing bit

OlivierC-FR, 2022-12-01:
I made a quick build with FC identifier as BTFL, not only for the DJI FPV side, just in case it is queried by a new feature, maybe the arming detection ? don't know if that was already done like that before ? I only have DJI since 1 week. On the theory the O3 finds the FC isn't BTFL before the OSD isn't even initialized, cutting grass under the feet of the "BF hack".

Anyway, didn't work because the INAV configurator way too cautious then says it wants nothing to do with this "BF" target, and I didn't have time to workaround and retry. This week-end I'll put more efforts into that.

jasc76, 2022-12-02:
This issue has tons of comments in a real short time, so a lot are interested. I think this is the first conclusion

the second thing: why are some saying they are actively blocking INAV. They said to a staff: go implement this OSD thing. So this person went and looked at BF and did it with the most basic knowledge and effort. This is not blocking, its just how things go. They made it work with BF. Imho this is not blocking its just incompatible. 
Is this good, of course not. 

But based on the interest, at least implementing some basic OSD, I think, would be appreciated by a lot of users. 

does anyone know rhis guy Chris Rosser talked to? Was he German? Maybe try to establish some sort of basic communication channel through him !?

b14ckyy, 2022-12-02:
@jasc76 not sure why you think that many comments show actual interest :D 

anyway. No it is not just this bit alone. As DJI has now implemented some kindof limited char set (font file if you will) this does not match the INAV char set at all. So this also has to be implemented now. Also there won't be a further INAV 6.0 test release before end of January at all. And the Final release end of march. That is 4 months from now where no Pilot can actually properly use this O3 air unit unless the pilots are willing to fly an unstable firmware.... IF the inav developers are willing to implement these changes at all. 

But nevertheless this whole thing just shows how little DJI cares. But I don't wanna repeat myself.

deethee, 2022-12-02:
You should definitely patch the latest release asap with something partially working and not wait till end January, cause the o3 is expensive and will go than in a betaflight kwads.

b14ckyy, 2022-12-02:
I don't patch anything I am not a developer. A new release is a lot of work and I highly doubt that anyone will release a 5.2 just because a handful of DJI Pilots jumped on the hype train too early and are now sitting on not usable hardware without checking compatibility first. Sorry mate :D

jasc76, 2022-12-02:
> @jasc76 not sure why you think that many comments show actual interest :D
>
> anyway. No it is not just this bit alone. As DJI has now implemented some kindof limited char set (font file if you will) [...]

But the discussion about how much they care doesn't lead us anywhere. Yes there is a limited font set. But at least there is any !!!! Thats a lot more then previously. So why not take advantage of what is there? 
Yes they could have interacted, I feel the same way like you, but from a user perspective: a limited OSD (at least also the menu works) is better than no OSD imho

claudiozavagno, 2022-12-02:
I have reconnected the old air unit which i use with dji goggles v1, but kept the O3 wiring in case something happens, or someone asks to do some testing. In the meantime the O3 will seat on the shelf. Bought them for the extra range which i only find it interesting in fixed wing use, where the only platform i am willing to use is Inav. Even a basic OSD (home pointer, distance and height) will take the O3 back in the plane and the goggles 2 on my face :-)

b14ckyy, 2022-12-02:
I know what you mean and yes from a user perspective it is bettr than nothing. but look at it from a "political" perspective. 
now the open sourse devs have to kneel down again and do the work for DJI.... again. DJI knows that and they don't care. This just shows them that it is okay to act like this. They save money on development that is then done by free open source developers. 

They throw stuff on the market in a sink or swim attitude because they already expect that we will sort it out ourselves. And by this the developers just show them that they can get away with it. 

Not sure if you remember the time when the FrSky Pixel OSD came to the market. They did something similar but back then the developers of betaflight started a bitch fight with them and refused to support it without seeing money for their work. FrSky got a shitstorm for that. DJI is praised for the same behavior. I don't get it and I will not support it.

MrD-RC, 2022-12-02:
> You should definitely patch the latest release asap with something partially working and not wait till end January, cause the o3 is expensive and will go than in a betaflight kwads.

You know demanding that something gets done is the quickest way to make developers (who are coding INAV in their free time, without pay) think "fuck that". Especially when it is adding support for a product, who's last version gave zero support to the community; and this new version is just doing more of the same. 

Stop demanding from INAV, and start demanding from DJI. They are the people who get your money, yet offer **no** support to the community. 

This new release of goggles 2 and air unit 03 was the perfect time for them to fix **their** mistakes from last time. But they have taken the exact same approach. Stop giving DJI your money. Use a system where the company does work with the community.

Jetrell, 2022-12-02:
From what I remember. DJI originally sent the INAV lead developer at the time. A V1 Goggle and Air Unit. So he could modify INAV to work with betaflight's MSP.

This only occurred after many INAV users emailed DJI. And asked them to support INAV OSD.

The only other option. Is if someone who can code, and owns a Goggle 2 / 03AU, comes along and writes a PR for its implementation.

MrD-RC, 2022-12-02:
> This only occurred after many INAV users emailed DJI. And asked them to support INAV OSD.

You'd have hoped that would have been a learning experience for DJI. Unfortunately not.

jalaninja, 2022-12-02:
I think it's now the right time to sell DJI gears. They're now hot in the market after O3 release. Walksnail seems like a better option, or maybe HDzero.

OlivierC-FR, 2022-12-02:
I also don't think that's a voluntary block. Just they made it work with BF and stopped there because its who they send the preprod units to. Minor uncompatibility is most likely.

a1axx, 2022-12-02:
@b14ckyy @MrD-RC Inav isn’t on their radar, unlikely to ever be as it’s a product aimed at DIY betaflight users. Why would they task anyone to go and make sure OSD works with inavs particular implementation is not compatible. Other FC firmwares are already compatible as they have available some kind of BF compatibility that works.

Fixed wing pilots that want osd can go and use ardupilot now. For the meantime it looks like the same question is going to be asked about o3 and OSD for a while. 

The German guy is Ferdi, Marc.

deethee, 2022-12-02:
I don't get it. Betaflight works out of the box, ardupilot works out of the box. Inav won't but someone already shown a way to partially fix the problem so that the o3 Air unit becomes usable which isn't now. And instead of making a minor release within 3h people start complaining about not being paid and about DJI, suggesting to sell your gear and buy hdzero, wow.. it is not just five people who jumped the hype train. 50%+ users are DJI and everyone still likes this hobby is exited about o3 and they will not come back if he needs to steps into ardupilot now.

b14ckyy, 2022-12-02:
Betaflight works because DJI wants it to work
Ardu works because it baqsically uses Betaflight CHar set to 90%
INAV does not work because the INAV Character Map is compeltely different. Here a comparison. All in red will be the wrong or unavailable icons even if the fix is applied. It is WAY MORE work for INAV to get a somewhat usable OSD working. 

[image]

deethee, 2022-12-02:
There are only a fraction of people flying wing compared to kwads and you really ask yourself why DJI is not adopting inav Fonts, while it would have been easily possible to stick to betaflight font and extending it i stead of brewing your own stuff?

MrD-RC, 2022-12-02:
> There are only a fraction of people flying wing compared to kwads and you really ask yourself why DJI is not adopting inav Fonts, while it would have been easily possible to stick to betaflight font and extending it i stead of brewing your own stuff?

Any data to support that rubbish 🤣 BTW, BetaFlight's font is changing in 4.4. So good luck with DJI then. Even the BF devs are saying DJI should have implemented MSP DisplayPort, like all the other HD systems. Including WTF, who fixed DJI's original crap OSD.

MrD-RC, 2022-12-02:
> I don't get it. Betaflight works out of the box, ardupilot works out of the box. 

BetaFlight works, because DJI did the same as they did last time.
ArduPilot OSD does **not** work on 03. ArduPilot is faking a BetaFlight OSD to get this working. But are missing OSD elements because of it. ArduPilot is also pretty big, so can't run on F411 or F722 flight controllers. They're both pretty popular fc's, especially for quads. 

> Inav won't but someone already shown a way to partially fix the problem so that the o3 Air unit becomes usable which isn't now. 

If that is all that's required to get DJI working with the shitty old DJI OSD. Then I'm sure it will be added. But the release schedule will not be brought forward for it. 

> And instead of making a minor release within 3h people start complaining about not being paid and about DJI, suggesting to sell your gear and buy hdzero, wow.. it is not just five people who jumped [...]

You are completely missing the point. It is not about INAV developers not being paid by DJI. No one has said that. What has been said is that DJI are the one who take your money, but don't give a shit about the pilots who buy it's products. The only way to try and make future DJI products compatible out the box it to have them interact with the development teams. They will only do this if their customers complain that their products don't work!  

Again. Where do you get the 50% of FPV pilots use DJI. I would say that is an overestimate by some way.

b14ckyy, 2022-12-02:
> There are only a fraction of people flying wing compared to kwads and you really ask yourself why DJI is not adopting inav Fonts, while it would have been easily possible to stick to betaflight font and extending it i stead of brewing your own stuff?

If it is only an unimportant small fraction of users who use INAV on fixed wings then why bother at all? In this case we can just ignore it and leave INAV as it is. Why developing something for a small fraction of people at all.

brousselle, 2022-12-02:
Here in the french Alps we are small group a flying wings (flying on inav, around 10 people) and the majority fly with DJI.
Seeing the number of comments in that issue and in telegram shows there is some interest in that feature. I don't think this fraction of users is that small. feature has also been somehow implemented and requested on wtfos, so there is a demand ;)

joshuabardwell, 2022-12-03:
> The very root of the issue is that they always send the first batch of gear to the same dozen of influencers, half of them make a video for the $ then put it on a shelf and never use it again, the other half fit on a BF quad and do like 3 flights "wooaaaaa product is awesome !!!" and that's the end of it. They get near zero valuable feedback from these people (excepted few I can count on a single hand) If only they would send one or two samples to INAV and arduplane users or devs, even the worse moron would think to send them an email like "eh what about adding support for everything else than a BF quad"... same sad old story, again. Sorry for the rant, they never learn.

This isn't quite right. They send us the gear. We beat the shit out of it and give them feedback. Then they (mostly but not entirely) ignore the feedback. Because by the time they send "influencers" the gear, 99% of the feature set is locked and they are in the final month or two leading up to release. All their testing has been done internally. They take our input and sometimes it has effects, but as far as, "Hey you guys need to support iNav," that needed to happen six months or a year ago, before anybody outside DJI even saw the gear.

OlivierC-FR, 2022-12-03:
> > The very root of the issue is that they always send the first batch of gear to the same dozen of influencers, half of them make a video for the $ then put it on a shelf and never use it again, the other half fit on a BF quad and do like 3 flights "wooaaaaa product is awesome !!!" and that's the end of it. They get near zero valuable feedback from these people (excepted few I can count on a single hand) If only they would send one or two samples to INAV and arduplane users or devs, even the worse moron would think to send them an email like "eh what about adding support for everything else than a BF quad"... same sad old story, again. Sorry for the rant, they never learn.
> 
> This isn't quite right. They send us the gear. We beat the shit out of it and give them feedback. Then they (mostly but not entirely) ignore the feedback. Because by the time they send "influencers" the gear, 99% of the feature set is locked and they are in the final month or two leading up to release. All their testing has been done internally. They take our input and sometimes it has effects, but as far as, "Hey you guys need to support iNav," that needed to happen six months or a year ago, before anybody outside DJI even saw the gear. The best thing we "influencers" can do is roast them in a video repeatedly, and then you can see how that goes by looking at how the development of Betaflight OSD support went. Here we are, 3 years later and we finally got it in the next generation of product.

Fair, but still i think half the gear they pre-send is wasted and only 2 units to INAV and Arduplane heads would make a huge difference. The actual compatibility issue with the OSD is not something that would require a 6 monthes back and forth cycle, I have zero doubt its a very minor detail that would be fixed in a single morning after a quick chat between a DJI coder and someone from INAV/AP. 

For what its worth, you are on my single hand of people I 
count who provide valuable feedback and move the whole community forward, thanks for all your hard work :)

donne1226, 2022-12-03:
I looked through all comments, and I cancelled my order!  still use my v1 goggles with wtfos. wont buy it until they respond to our community.
I also created a thread in dji official forum(CN). let's wait and see what happens.

jalaninja, 2022-12-03:
> I looked through all comments, and I cancelled my order! still use my v1 goggles with wtfos. wont buy it until they respond to our community. I also created a thread in dji official forum(CN). let's wait and see what happens.

I Installed WTF-OS last night after reading this discussion 😂
Risking vista/link set + runcam/caddx 4k camera is still cheaper than O3

erstec, 2022-12-03:
@deethee you asking for "fix" without knowing how to program? Srsly? Why not to do this by yourself if you know that it is "simple" and will not took time? Do that, make PR and we (community) will say you very big Thank you for that! Whatever...

We are NOT develop without having real equipment on a bench. It is impossible to made "working well" functionality just based on someones "tests". It will be possible ONLY when some of us will decide to spend 1k+ of "money" to buy set of DJI V2 googles and at least one O3 airunit from our own cash, or... DJI will provide samples to core devs.

And yes, if you want to have it RIGHT NOW, consider to hire developer (don't need to be from INAV team). There is no obligations from team at all to do anything or not. Every one can contribute. Everyone. Read licence agreement more carefully before claiming things.

Claiming here will not lead anything except menthal block to your claims.

MrD-RC, 2022-12-03:
This may help anyone looking to jump on this. The symbols have been match to BetaFlight/DJI O3. Obviously not tested, as I don't have the system. But if `USE_DJI_O3_OSD` is defined, it will swap the symbol indexes. So current OSD drawing functions should work without change. It will of course need the DisplayPort stuff all done, and configurator will need to be limited to only show the OSD elements that will still work.

https://github.com/iNavFlight/inav/tree/MrD_Add-defines-for-the-BF-OSD-charset

jasc76, 2022-12-03:
I got also some additional info:WS and HDZ both implemented the ability to define, upload and maintain font set in the goggles. So in fact just a file on the goggle. The goggle then selects the font file based on the MSP data. DJI just again only did the bare minimum and baked some font set into their firmware. If they asked someone how to do it, we would nit have this shitshow again.I ordered DJI, against all my feelings and got badly disappointed now. I was in the assumption DJI has MSP Canvas/Displayport whatever and I can use it with copters and wings. But now …. I‘m left high and dry. I missed the WS sale, cant get the HDZ goggle and just feel miserableRegarddsJan

b14ckyy, 2022-12-03:
So Pawel made it official right now. INAV will NOT build any complex workaround to make O3 Goggle 2 OSD work with INAV. Not again. Either DJI gets their stuff sorted and understands that there are customers they miss out or you can't have OSD on INAV. 

Send them mails, post in their forum. Flood them with requests and see if that helps.

a1axx, 2022-12-03:
> So Pawel made it official right now. INAV will NOT build any complex workaround to make O3 and Goggle 2 OSD work with INAV. Not again. Either DJI gets their stuff sorted and understands that there are customers they miss out or you can't have OSD on INAV.
> 
> Send them mails, post in their forum. Flood them with requests and see if that helps.

I think he did mention a basic work around might be possible. D Lines has made a start on that.

b14ckyy, 2022-12-03:
@a1axx you misunderstood that. There is a basic workaround to get the message transfer to the VTX working yes. But that does not fix the completely wrong font IDs that would need much more work. Darren has just some testing if a workaround with limited OSD would be feasible storage wise for F722 but when Pawel says that INAV will not buckle again to make another half baked solution lilke their last OSD. 

DJI Broke all the compatibility to anything on their end and if they do not actively work with the developers and actively integrate INAV OSD support on their end, there will be no support.

[40 more replies]