iNavFlight/inav #8582: Allow remap motor and servo pins

xpol, 2022-11-25:
I want to use Matek H743 mini (the one designed for drones) for planes.
But according to its layout, the S1-S4 are in a JST 8pin socket which is not ideal for servos.
[image]

So I'd like to use, S1~S4 for motors (maybe only S1-2, and leave S3 and S4 free) and S5-S8 for servos.

## Current Behavior
But the current iNav only gives me the following mapping:
[image]

## Desired Behavior
And What I wanted:
[image]

## Suggested Solution

- Can I change the iNav/Configurator code to implement this? Where should I start?
- Feature #1145 should also be a solution for this.

## Who does this impact? Who is this for?

Airplane users
Users of rich IO flight controllers, like Matek H743 mini, will need this.

## Additional context

closed 2022-11-25 by MrD-RC

breadoven, 2022-11-25:
Surely you can still add servo mixers for S7 and S8 allowing the use of the through hole pin connections for S5 to S8,. Then set up as required, roll, pitch, yaw etc. Servo mixers S3 and S4 just don't get used, Not remapping but should still work so long as you don't need more than 2 motors.

xpol, 2022-11-28:
Got what I want, just replace the `TIM_USE_FW_SERVO` with `TIM_USE_FW_MOTOR`.
For me I changed for on *S3* and *S4* in the [here](https://github.com/iNavFlight/inav/blob/master/src/main/target/MATEKH743/target.c#L38-L39).