iNavFlight/inav #8594: Current readings in OSD Matek f722 WPX when on 6S battery

jayss66, 2022-11-29:
Matek f722 wpx INAV 6.0  Current readings in osd only rise to around 3.2amps then as throttle is raised further the current values start going down. Works fine on 4S battery.
Any ideas?

b14ckyy, 2022-11-30:
do you have a capacitor added to your ESC or ESC pads of the FC? (Not on the main power lead) 
Could be electric noise that interferes with the current reading ADC.

dingrgh, 2023-03-23:
> Matek f722 wpx INAV 6.0 Current readings in osd only rise to around 3.2amps then as throttle is raised further the current values start going down. Works fine on 4S battery.
> Any ideas?

I had a similar problem with my Matek F722-WPX together with a BLHeli_32 Tekko32. Wild current readings at mid-throttle. It was fixed with a 1000µF capacitor across the ESC power pads. As a point to note, a BLHeli_32 ESC of a different brand did not produce the same problem with the same FC.

closed 2023-10-27 by DzikuVx