iNavFlight/inav #8595: ESCs will Not give second beep in INAV but works fine in betaflight and Blhelisuite

PATM5771, 2022-11-30:
So strange I have everything hooked up properly but battery will not give second beep and cant get motors to spin in INAV.  i switched back to betaflight and it works fine Hobbywing 60a ESC Mamba MK@ (FURYF4OSD) Checked the escs in blhelisuite they all read fine and even updated to the latest firmware.  in INAV I cant even get them to spin in the outputs tab in INAV.  I checked the resources on both inav and betaflight and motors are all same pins.  any idea? Im stumped

b14ckyy, 2022-11-30:
1. have you enabled servo and motor ouptuts in Outputs tab?
2. What motor Protocol did you set up? Some modern BLHeli ESCs don't support STANDARD PWM anymore and need DShot
3. If STANDARD is supported and set, have you calibrated the ESC? 
4. What Protocol was set on Betaflight? 
5. Did you flash the correct INAV Target to your FC? What exact FC model? 
6. What INAV version?

PATM5771, 2022-11-30:
All dshot each type I’ve tried. Betaflight was shot also. Tried inav 5.0
and 5.1 and loaded correct mixers. Target is furyf4osd but it’s a Namba
f405 mk2 which makes you use the fury target for some reason in their
documentation. I was gonna try a matek f7 fc next and see if there’s a dif
because inav not recognizing the LiDAR sensor either.  But check motor pins
in resources on both inav and betaflight and motor pins are at least the
same. Only thought I’d it’s the fc because getting two dif problems for no
reason. Thanks for your help and input

PATM5771, 2022-11-30:
Esc hobbywing 60a is recognized in blhelisuite without the second beep and
shows all normal info  also.

b14ckyy, 2022-11-30:
The Mamba FCs are a nightmare regarding targets for specific versions. I have one in my 5" racer but I think I used nopt the fury target actually for BF. need to check what version and if INAV works on there. If I don't forget it I will check later.

PATM5771, 2022-11-30:
Ok, thanks.  Appreciate it. I’m gonna try a matek f7 std and see if that
works because I’m getting lights on the LiDAR sensor but that still doesn’t
work either.

sensei-hacker, 2022-12-05:
I have an idea of what the problem may be, and a possible solution.
can you please provide a link to the exact ESC you're using, or as close as you can find?

You mentioned it worked in Betaflight.  Was that using Standard protocol, or dhsot or multishot?

PATM5771, 2022-12-05:
Hobbywing 60a 4 in 1 and dshot600. I switched to a different fc and esc
works in inav but LiDAR still won’t. It lights up but inav can’t detect I
think the LiDAR is faulty

sensei-hacker, 2022-12-05:
Okay the 4-in-1 using DSHOT 600 would be a different issue than what I was thinking of.

That said, I would try DSHOT 300 or 150 next time if you mess with that ESC again. Lower bit rates are less sensitive to issues.

closed 2024-03-22 by b14ckyy