iNavFlight/inav #8596: MSP display-port does not work with the new DJI O3 air units

OlivierC-FR, 2022-11-30:
Gear : 
DJI Goggles V2, not rooted, not WTFOS modified, with newest official DJI firmware (v1.04.0000 2022-11-15)
DJI O3 air unit, with newest official DJI firmware (v1.01.0000 2022-11-15)
Matek F722-SE
INAV 6 FP1
INAV configurator FP2

O3 AU is connected to the UART, peripheral = MSP displayport, the OSD does not show in the V2 goggles.
Can see a basic link is up tho, the O3 detects the arming of the motors via MSP, that is then used to auto-trigger the recording and auto switch out of the low power state.

----

3 screencap below are from Betaflight latest version, 4.3.2, the OSD works in the V2 by just checking MSP for the UART. 

[image]
[image]

[image]

deethee, 2022-12-01:
Can confirm this. Flying without OSD on a wing makes this o3 air unit worthless.

a1axx, 2022-12-01:
Here is the compatible character set 
[image]

a1axx, 2022-12-01:
Passed on from a user on discord:

[image]

MrD-RC, 2022-12-01:
That character set is pretty much useless for a full INAV OSD. Looks like DJI have done it again.

deethee, 2022-12-01:
Could you describe how you add the heartbeat or link the more detailed Information to that discord?

MrD-RC, 2022-12-01:
I'm going to close this issue, as it is basically the same problem with issue https://github.com/iNavFlight/inav/issues/8578 That problem being DJI.

closed 2022-12-01 by MrD-RC