iNavFlight/inav #8597: OSD elements randomly disappear in flight when using MPS/HDZero option & Walksnail Avatar video system.

JoeSa3rd, 2022-11-30:
## Current Behavior
During flight some OSD elements randomly disappear, of which never come back unless you power cycle the FC and Goggles. This happens at random in flight whether you are close in (E.g. 100-500ft) with clear LOS or flying further out. This same issue has happen on all 3 of my iNav/Avatar planes with various flight controller types all running iNav v5.1. I also have two other friends running iNav 5.0 and v5.1 with the same exact issue.

## Steps to Reproduce
1. Any FC Running iNav v5.1.
2. Avatar FW 28.32.10 on VTX with Nano or Micro Camera.
3. Configure required VTX UART port with MSP/HDZero options.
4. Fly to experience OSD elements randomly disappear.

## Expected behavior
OSD Elements do not disappear

## Suggested solution(s)
Investigate why MSP is not working well with the Avatar video system to help find a root cause.

## Additional context
Unfortunately I don't have any video captured of the issue as the Avatar goggles do not record the iNav OSD in flight.

Although I don't have any conclusive evidence I/we have noticed this issue more often when multiple people are flying the Avatar system at the same time but the issue still happens when flying alone but seems less often. again might not be related but wanted to share it.

---
FC Matek F405-WMN or F405-WTE or H743-WLITE
diff all of Matek F405-WMN

# version
# INAV/MATEKF405TE 5.1.0 Aug 19 2022 / 12:08:11 (76f22b25)
# GCC-10.2.1 20201103 (release)

# start the command batch
batch start

# reset configuration to default settings
defaults noreboot

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mixer
smix reset

smix 0 1 1 100 0 -1
smix 1 2 0 100 0 -1
smix 2 3 0 -100 0 -1

# servo
servo 1 1000 2000 1490 -100
servo 2 1000 2000 1490 -100
servo 3 1000 2000 1490 100

# safehome

# feature
feature -TX_PROF_SEL
feature MOTOR_STOP
feature GPS
feature LED_STRIP
feature PWM_OUTPUT_ENABLE
feature FW_LAUNCH
feature FW_AUTOTRIM

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS

# map
map TAER

# serial
serial 2 33554432 115200 115200 0 115200
serial 3 2 115200 115200 0 115200

# led
led 0 1,0::CB:2

# color

# mode_color

# aux
aux 0 0 0 1700 2100
aux 1 2 2 1700 2100
aux 2 12 2 900 1300
aux 3 45 6 1300 1700
aux 4 53 6 1700 2100
aux 5 11 5 1300 1700
aux 6 10 5 1700 2100
aux 7 21 4 1300 1700
aux 8 54 4 1700 2100
aux 9 54 2 1700 2100
aux 10 13 1 1300 2100
aux 11 47 3 900 1300

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 23 0 H
osd_layout 0 1 30 13 V
osd_layout 0 6 23 13 H
osd_layout 0 7 20 14 V
osd_layout 0 9 1 2 H
osd_layout 0 11 13 13 V
osd_layout 0 12 18 13 V
osd_layout 0 13 24 12 V
osd_layout 0 14 8 13 V
osd_layout 0 15 18 12 V
osd_layout 0 22 17 9 H
osd_layout 0 23 30 12 V
osd_layout 0 28 30 14 V
osd_layout 0 30 9 15 V
osd_layout 0 31 8 12 V
osd_layout 0 32 24 13 V
osd_layout 0 40 8 9 V
osd_layout 0 41 8 8 V
osd_layout 0 110 8 11 V
osd_layout 0 112 8 10 V

# logic

# global vars

# programmable pid controllers

# master
set gyro_main_lpf_hz = 25
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 30
set gyro_zero_x = 2
set gyro_zero_y = -1
set gyro_zero_z = -4
set ins_gravity_cmss =  980.894
set acc_hardware = ICM42605
set acczero_y = -6
set acczero_z = -2
set accgain_x = 4098
set accgain_z = 4090
set align_mag = CW270FLIP
set mag_hardware = NONE
set baro_hardware = SPL06
set motor_pwm_protocol = STANDARD
set failsafe_procedure = RTH
set current_meter_scale = 250
set platform_type = AIRPLANE
set model_preview_type = 14
set imu_dcm_ki = 0
set small_angle = 180
set imu_acc_ignore_rate = 7
set imu_acc_ignore_slope = 4
set applied_defaults = 3
set gps_sbas_mode = WAAS
set gps_ublox_use_galileo = ON
set airmode_type = STICK_CENTER_ONCE
set nav_wp_radius = 5000
set nav_rth_allow_landing = FS_ONLY
set nav_rth_alt_control_override = ON
set nav_rth_altitude = 5000
set nav_fw_control_smoothness = 2
set nav_fw_launch_max_angle = 180
set nav_fw_launch_motor_delay = 0
set nav_fw_launch_timeout = 45000
set nav_fw_launch_climb_angle = 20
set osd_video_system = HD
set osd_units = IMPERIAL
set osd_time_alarm = 30
set osd_alt_alarm = 9999
set osd_dist_alarm = 0
set osd_crsf_lq_format = TYPE2
set osd_main_voltage_decimals = 2
set tz_automatic_dst = USA

# profile
profile 1

set fw_p_pitch = 15
set fw_i_pitch = 15
set fw_d_pitch = 5
set fw_ff_pitch = 97
set fw_p_roll = 15
set fw_i_roll = 10
set fw_d_roll = 7
set fw_ff_roll = 41
set fw_p_yaw = 20
set fw_i_yaw = 0
set fw_ff_yaw = 100
set fw_p_level = 30
set max_angle_inclination_rll = 450
set dterm_lpf_hz = 10
set fw_turn_assist_pitch_gain =  0.400
set nav_fw_pos_z_p = 25
set nav_fw_pos_z_d = 8
set nav_fw_pos_xy_p = 55
set d_boost_min =  1.000
set d_boost_max =  1.000
set fw_level_pitch_trim = -2.156
set rc_expo = 35
set rc_yaw_expo = 30
set roll_rate = 26
set pitch_rate = 13
set yaw_rate = 3
set manual_roll_rate = 65
set manual_pitch_rate = 75

# profile
profile 2

# profile
profile 3

# battery_profile
battery_profile 1

set vbat_min_cell_voltage = 260
set vbat_warning_cell_voltage = 300
set battery_capacity = 3500
set throttle_idle =  5.000
set nav_fw_launch_idle_thr = 1300

# battery_profile
battery_profile 2

# battery_profile
battery_profile 3

# restore original profile selection
profile 1
battery_profile 1

# save configuration
save

#

MrD-RC, 2022-11-30:
Known issue being worked on.

JoeSa3rd, 2022-11-30:
Glad to hear this. If you need anyone to help test a possible fix, I'd be more than willing to help test.

maRCnh33, 2023-10-10:
I use inav6.1.1 (latest version) and i have also that problem. 2 items on the osd disappeared for about 5-10 seconds and then they come back again. The strange thing is that it is only 'ALTITUDE' and 'Flying time'..    so now on inav 6 the issue is still not fixed??? I use also msp display and hdzero in the osd tab. Can someone fix this problem??

MrD-RC, 2023-10-10:
INAV is doing everything it can now. It is really up to the HD systems to improve the robustness of the OSD data transmissions.

closed 2023-10-10 by MrD-RC