iNavFlight/inav #8603: Inav 3.0.0.1 WTFOSD Missing some OSD elements

maRCnh33, 2022-12-02:
I have just rooted my DJI FPV Goggles V2 and installed the wtfos system, also the packages.

Then i had to set the right port for it in inav, that went okay.

But now, the strange thing is that i get a lot of osd from inav but NOT ALL, after my first view in the goggles i saw that there was no 'Distance to home' and 'total flight distance' ..... also the 'horizon line' in the middle is missing.

Maybe i have to ask for help here or at WTFOS ...... But the wtfos osd is working at least but i miss the most important items now... -:(

My thought was that i can something has to do about the fonts... but.... other osd text is displayed!?!?

I like to get some help, meybe someone knows the issue and the solution to fix it.

Tnx

MrD-RC, 2022-12-03:
You need to be using at least INAV 4.1 for WTF. I recommend upgrading to the latest stable release, 5.1. 

Use the HDZero OSD peripheral for WTF.

maRCnh33, 2022-12-03:
Thank you very much for your quick response! I will do the firmware update!
Can i make a cli dumpfile and paste it in 5.1? Or maybe a diff export and
then import in 5.1?

Otherwise i will do a clean setup from scratch..

Tnx
Marc

MrD-RC, 2022-12-03:
Never use a **dump** for an update. They are only to restore settings on the same flight controller, in the same model, on the same version. To be honest, dumps are overkill in 99.9% of cases. 

I would recommend taking a **diff all** as a backup of your current setup. With regards on what you can use from the diff, it's always worth reading the release notes. They will reveal all. Off the top of my head:
- don't copy any of the profiles
- don't copy the **# master** section. Other than the three **accgain** and three **acczero** parameters

The rest of the diff _should_ be ok. But double check with the release notes.

maRCnh33, 2022-12-03:
Thanks for the advice! Today i will update fc to inav 5.1.

You made everything clear to me!

Have a nice day..

Marc

maRCnh33, 2022-12-03:
I just installed inav 5.1 on my quad. Now i am setting up everything. Now i
am on the ports tab and setting up the port for the DJI airunit. You said
to use HDZero osd but i am not sure now.... To be clear, its an DJI
airunit. Do i have to choose DJI FPV VTX or HDZero? Please let me know!
Tnx..... I like to know that the setting is 100% good -:)

MrD-RC, 2022-12-03:
For WTF use HDZero.

maRCnh33, 2022-12-03:
its working! very nice.. everything is done now, only the magnetometer
calibration i have to do (outside without any electronics -:) beside!
Finally the osd on screen that I WANT and not what is 'necessary' for DJI!
Quad is done now, 2 airplanes are left to do the same, update to INAV 5.1 .
A volantex Firstar 1600 and a Multiplex Twinstar ND (My latest build!) With
headtracker module and good pan/tilt servo's, wireless!! With all the osd
elements it feels more safe to fly, i can't wait to test the quad and
planes (planes i will do tomorrow and then everything is ready to fly! I
made a diff backup and i set everything manually and only for the switches
and the ledstrip i put back the items from the diff.....So nice and fresh
setup!! And again : *Thanks for your help!!*

closed 2022-12-03 by MrD-RC

mooiweertje, 2022-12-04:
geen dank

maRCnh33, 2022-12-04:
voor de duidelijkheid, ik ben je heel dankbaar dat je de firmware voor m'n
vliegtuigen hebt gedaan hoor! Ben nu bezig met inav 5.1 installeren op
vliegtuigen. Via configurator op pc steeds geklooi waardoor ik vanaf daar
niet kan flashen.. nu even de jnav5.1 naar mijn telefoon sturen en het van
daar met speedybee app proberen. En ik laat het je weten als het goed
werkt! Met vriendelijke groetjes Marc