iNavFlight/inav #8605: Zero looptime value problem

demvlad, 2022-12-04:
If i set looptime value to zero in cli (set looptime=0),
then INAV-configurator does not show configuration after connecting, and flight controller does not work. As write in help, valid value of looptime are 0....9000.

- FC Board name and vendor: Matek F411WSE
- INAV version string: 5.1, 6.0

stronnag, 2022-12-04:
Don't do that then. 
The CLI comes with no guarantee to protect from stupidity.

stronnag, 2022-12-04:
It is also documented behaviour.
https://github.com/iNavFlight/inav/blob/master/docs/Settings.md#looptime

closed 2022-12-04 by DzikuVx

demvlad, 2022-12-04:
Thank you for answer.
I have read this information before. It is wrote : "Setting it to zero does 
not limit loop time, so it will go as fast as possible..."..
I thought this will work as described in this document... Ok, never mind, i 
will not do this in future...