iNavFlight/inav #8607: Support for 1,3 Zoll OLED Display I2C SSH1106

b14ckyy, 2022-12-05:
## Current Behavior
SSH1106 OLED Displays show content but the alignment of the text is incorrect (slighly shifted off screen to the left) and the screen is not correctly refreshed. Shows Garbage instead of black area and does not clear when entering the CMS. 

[image]
[image]

## Desired Behavior
Fully functional SSH1106 1.3" OLED Displays.

## Suggested Solution
Implement SSH1106 Drivers

## Who does this impact? Who is this for?
Any Fixed wing pilot who likes to use OLEDs for field checks and setting changes in the field without the need of a running VTX. Especially Digital FPV users where VTX overheat even at 25mW. SSD1306 OLEDs work great but are very small. Especially on bigger crafts it might be nice to have a bigger 1.3" OLED display instead, if the room is there. 

## Additional context
- I have a few of these SSH1106 and can send one to any developer in EU if needed.

closed 2023-10-27 by DzikuVx

alexandre-mazel, 2023-12-23:
Hello,
I have quite the same problem of garbage pixel on the right of the screen. By chance I figure out the width is perhaps different, so I try different settings, and after complicated working things, I find this simpler solution that works fine:

// add in enum in oled.h:
    /** Possible display width values **/
    enum tWidth { W_96=96, W_128=128, W_132=132 };

// call to construct object:
OLED disp(20,21,NO_RESET_PIN,OLED::W_132,OLED::H_64,OLED::CTRL_SH1106,0x3C);

That's all !

alexandre-mazel, 2023-12-23:
And to correct the fact pixels are offset on the left on the screen i add this in the setup:
disp.useOffset(1);

BainesBunch, 2024-12-18:
DId this ever get into the v8.0.0 release, I have one of these and am still seeing the same issues with the display.

svpetry, 2025-03-13:
I am also having this issue with 8.0.0.