iNavFlight/inav #8609: Telemetry form FC through Flysky sensors I-bus not working.

Outwide, 2022-12-06:
## Current Behavior
Not able to get Telemetry from FC when setting telemetry to I-Bus out and into the connection port of Flysky sensors I-bus. Only the Flysky sensors outputs with NO FC telemetry. 
If you try Flysky sensors I-bus and connect to a UART of the FC and set Telemetry out I-Bus out only the FC telemetry outputs with NO Flysky sensors output. 

## Desired Behavior
To be able to use all Telemetry form both FC and Flysky sensors, I-Bus protocol out to Flysky receiver telemetry capable I-Bus. 

## Suggested Solution
1. To be able to set either by the CLI or GUI tabs settings
2. To be able to pass I-Bus Telemetry out form FC and into the connection port of any Flysky sensors and output telemetry to a Flysky receiver telemetry capable I-Bus. AFHDS3, AFHDS 2A 

## Who does this impact? Who is this for?
Those who don't use blheli_32 esc telemetry and want more telemetry and flexibility in the software. 

## Additional context
Being able to add extra telemetry to Inav that is NOT provided, such as RPM, TEMP and Volts for bigger motor esc with no telemetry support. This can only be better for the sport and hobby.

Outwide, 2022-12-11:
I ask is it possible to add extra RPM sensor that is attached to the wires of the motor. This maybe be able to be added in the line 213 - as seen in the (https://github.com/iNavFlight/inav/tree/master/src/main/target/MATEKH743)/target.h
[image]

closed 2024-03-22 by b14ckyy

DantesInferi, 2024-05-25:
Did you connect Tx to Rx on SENS? Or just bi-directional?
Did you ever get it to work?

Outwide, 2024-05-25:
I connected to the flight controller rx as I wanted use the all senors that I had or believed where compatible, both on the flight controller and what could buy. I have not been able to check if it has been fixed as indicated to me. I wanted , temp , rpm sensors.