iNavFlight/inav #8612: Inav5.1 - Flaps ch12, MANUAL is also on that switch and flaps goes up while using MANUAL

maRCnh33, 2022-12-06:
I have inav5.1 installed on my rc airplane. Because of the missing 'has flaps' function i set up the flaperons on channel 12. 

CH 12 is on a 3-way switch, TOP position (1500us) = nothing, MIDDLE position (2000us) is flaps and BOTTOM position (1000us) is MANUAL. 

In the mixer tab i added 2 more channels for FLAPERONS, on the same servo's as AIL L & R.

Now i have the issue that the ailerons goes up (spoileron) when i select MANUAL.

How to fix this so that the FLAPERONS are not working at 1000us...

I think it must be done in the 'programming' tab but i really don't no how -:)

Maybe someone has a solution? Tnx

Marc

i do not need spoilerons, only flaperons ...

MrD-RC, 2022-12-06:
The hadFlaps switch basically did nothing. It just enabled the flaperons mode on the modes page. This is now automatic, if the mixer you choose is suitable. 

Just set up mixes for flaperons mode. And put the switch for the mode where you want it. 

To fix your problem, move the slider for the flaperons mode. To be honest. There are better ways of setting up flaperons than using that mode.

maRCnh33, 2022-12-09:
its fixed! The problem was that I had no 'free' switch..... so i used a
3-position switch but that didn't work... (because it had several functions
on 1 switch...)

Your video has explained a lot to me thanks!

What i have done is changed a potentiometer with a 2 position switch (with
2 resistors on the switch and that will be the 'flaperon' switch...
(because that turn knob i never in use and i have 2 so i have 1 knob to use
later maybe.... )

Tnx

Happy flying!

Best Regards Marc

fpvMarkS, 2023-09-08:
so in the new INAV (6.1 horizon hawk) 
flaperons in the modes tab does nothing but make the FC beep when enabling flaperons?
Is that correct ???

MrD-RC, 2023-09-08:
No. It’s the same as 5.1. It hasn’t changed.

closed 2023-09-08 by MrD-RC