iNavFlight/inav #8613: Battery failsafe

sdellava, 2022-12-06:
## Current Behavior
At the moment the there is no battery failsafe.

## Desired Behavior
Let the FC take care of the remaining energy in the battery and enter in RTH mode in case the level drop below a defined limit.

## Suggested Solution
A specific configuration allows you to define whether to activate RTH in case the battery residual energy level falls below a limit.

The limit can be set manually, with hint:
- 30% if RTH is set for a direct flight to home
- 50% if the RTH is set for a flight along the path taken (trackback).

## Who does this impact? Who is this for?
All users. 

This feature benefits both beginners users who may not know how to correctly estimate the amount of power needed to safely return to base, and it also benefits experienced users who may nevertheless have an unexpected problem with faulty or underperforming batteries due to weather conditions. Confident that you have enough battery power, it is easy to miss the fact that you are actually running the risk of not re-entering.

The battery failsafe feature could also be used to disable a planned mission and reenter the aircraft.

kasatka60, 2022-12-06:
Usually, the voltage value and the consumed value flash on the osd so that the pilot pays attention and returns home.

b14ckyy, 2022-12-06:
Are you referring to Fixed Wing or quads? or both? 
I forced RTH based on battery level is pretty much useless. What if you fly just a mission close to you? If you want to override it, you would need an extra channel for it. 

For fixed wings we already have an estimated remaining flight time and remaining flight distance before RTH has to be initiated to make it home. This takes distance, cruise speed and wind speed into account. On a quad this is much more difficult as we can barely accurately measure wind speed to account for it. 
also this feature needs a lot of testing and fine tune for every single craft. 

Pure battery percentage-based return is no option, and a flight time/distance option is hard to do for quads. Not even DJI takes wind into account. just distance, and time based on max navigation speed.

sdellava, 2022-12-07:
Thank you @b14ckyy for your input. The programming option is something I hadn't thought of and will try, but it is not an alternative.

I think we can divide this interesting discussion into three topics:
1) Is the battery-powered RTH procedure a useful function? 
2) How to assess when the situation requires the RTH procedure to be activated?
3) How to exclude the procedure.

Point 1: This is an optional configuration, so each user is free to use it or not according to his or her personal perception of the benefit he gets. 

Point 2: We can define many different approaches to determine when to activate automatic RTH. I used the word "battery" because it makes the purpose clear, but other inputs can be considered, such as an "estimate of the energy cost of the route home" taking into account, if possible, the wind, or the change in elevation of the route home, etc. 

Wind speed can only be estimated with specific sensors, but the energy to go back along the route taken on the way out following the change in altitude in the route, can be estimated without sensors and this estimation is not so easy to do by the pilot. Other methods can be considered and of course the fixed percentage of battery left.

The thought I started from to make this suggestion is that it is better to land somewhere close to home instead of landing far away because the pilot miscalculated the capacity of the aircraft.  

Point 3: The RTH procedure should be able to be modified by the pilot. For example, the pilot can explicitly activate the RTH mode and then have control of the aircraft as already planned.

synersignart, 2023-03-17:
I think Battery failsafe can be set into Programming . when Voltage reach a set value . it puts the drone flight mode into RTH. that simple

MrD-RC, 2023-03-17:
I agree with @b14ckyy and @synersignart 

This can be done in the programming framework. You can set a voltage to activate RTH. You can even add a way you exit it; if you enter a mode (acro for example, something you're unlikely to be using when cruising far away), and the home distance doesn't increase, you have control again. Once you have control again, you can use any mode. Acro is just the trigger.

closed 2023-03-17 by sdellava