iNavFlight/inav #8618: Damaged on-board Barometer, adding external Barometer is not discoverable

roligmue, 2022-12-07:
Summary question: 
Is there a way to use, or map, an external Barometer in iNav, if their was an onboard Barometer that is no longer working, or has since been removed? 

Late night build error on my part - I accidentally permanently damaged a iFlight F7 TwinG onboard barometer by conformal coating the chip. It had been working up to that point. I have since removed the barometer chip by carefully breaking it off and chipping off its remnants. I have used this external barometer on multiple Betaflight builds and a iNav build too, using a GY-BMP280 Module on each. It is wired in parallell to the SCL SDA ports and powered by 3.3v on the flight controller.

[image]
In iNav, the i2c bus is actively working as compass is active with telemetry working, however barometer is red in configurator. Barometer is set at its default address 0x76. It can be changed to 0x77 by soldering an additional wire, but that did not work either. 
i2c errors do not increase after booting flight controller.

CLI Status:
status
System Uptime: 1151 seconds
Current Time: 2022-12-07T15:11:45.568+00:00
Voltage: 0.04V (1S battery - NOT PRESENT)
CPU Clock=216MHz, GYRO=MPU6500, ACC=MPU6500, MAG=QMC5883
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=UNAVAILABLE, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK, IMU2=NONE
Stack size: 6144, Stack address: 0x20010000, Heap available: 1924
I2C Errors: 50, config size: 9658, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 11, cycle time: 504, PID rate: 1984, RX rate: 49, System rate: 9
Arming disabled flags: HWFAIL RX CLI DSHOTBEEPER
VTX: band: A, chan: 2, power: 1 (25 mW), freq: 5845 MHz

[INAV_5.1.0_cli_dump bad barometer_20221206_144546.txt](https://github.com/iNavFlight/inav/files/10177278/INAV_5.1.0_cli_dump.bad.barometer_20221206_144546.txt)

Seems to be an address configure issue to be able to communicate to this external barometer.

I can try loading Betaflight and trying its configure settings, but wanted to check here first.

Thank you

b14ckyy, 2022-12-09:
where did you solder the new baro to? I see no I2C pads on that FC: 
[image]
[image]

Yes I know that's the version without baro but the images where better for that one.

roligmue, 2022-12-09:
Here’s the pads for this Board.Rick OligmuellerMobile 612-963-0290

b14ckyy, 2022-12-09:
@roligmue sorry mate, mail answers don't include pictures.

closed 2023-04-18 by DzikuVx