iNavFlight/inav #8619: nav_extra_arming_safety = ALLOW_BYPASS does not allow Arming without GPS Lock

ivanlimwk, 2022-12-08:
In iNav 5.1 nav_extra_arming_safety = off works. With new iNav 6 RC1 nav_extra_arming_safety = ALLOW_BYPASS is only option available. I had to have 6 Sat before I could arm, even i have it set as Allow_Bypass in CLI

ivanlimwk, 2022-12-08:
"It only lasts until the flight controller is rebooted." Meaning I cannot reboot the FC after I have set nav_extra_arming_safety = ALLOW_BYPASS in CLI? Am I understanding this right. Doesn't the FC reboot when i save the CLI?

stronnag, 2022-12-08:
Once set in the CLI, `ALLOW_BYPASS` does what is says. It allows you to arm (with right raw while activating arm) without meeting navigation safety requirements.
If you subsequently disarm, you can arm normally (without right yaw) until reboot, when the bypass procedure is again required if arming safety conditions are not yet.

ivanlimwk, 2022-12-08:
I got it now. I was not aware of the Right Yaw stick to arm as a bypass. in 5.1 i just armed. Thank you for the Help guys

closed 2022-12-08 by ivanlimwk