iNavFlight/inav #8623: [DRAFT] Concept for fully autonomous Safehome + Waypoint Landings for FW

b14ckyy, 2022-12-11:
Thanks to all the awesome new features implemented in INAV for 6.0, it now has the necessary Navigation Precision and capabilities, to allow fully autonomous landing with linear approach and soft touchdown. Because of that I worked on a concept for a capable Autoland feature that can kick in for failsafe landings and even autonomous missions with outside landings (even one-way missions). The basic idea is, to automatically create waypoints for the landing approach, based on predefined landing vectors and approach parameters. 

Needed Sensors:
- Accellerometer
- Gyro
- GPS
- Barometer
- Compass (recommended)
- Lidar/Sonar Rangefinder (highly recommended)

### Process: 
**Phase 1 - Direction Decision**
- Before the landing approach begins, INAV loiters at the target position for at least 30s to properly determine the wind direction and speed and make sure the data are up2date and precise. Then INAV decides about landing direction (details see below - Landing Spot Parameters)
- Then the plane flies to the automatically created approach start waypoint. This will be defined by `nav_fw_land_appr_length` (orange line) and `nav_fw_land_appr_alt`. 
- from here on, Waypoint path tracking  and turn smoothing in CUT mode needs to be forced on with default values (5 and 45°), if not already enabled and tuned by the user
- as soon as the approach start point is reached, Phase 2 Approach starts

**Phase 2 - Approach Alignment**
- During the Approach phase, a normal waypoint track is flown, not different to a WP mission
- the target waypoint is automatically generated in the same direction as the approach start waypoint and defined by `nav_fw_land_glide_length` and `nav_fw_land_glide_alt`
- **(to discuss)** During the approach phase, users should be able to set certain channels automatically. Ideal would be to have Logic Condition Triggers. One General Trigger during the whole landing Phase should be enough as additional triggers can be set by the user, like altitude). This can be used to extend flaps 50% and Landing Lights when the approach starts and then go Flaps full and extend landing gear at a certain altitude. User can do that easily in the logic conditions afterwards. 
- The Approach Phase is basically a track to allow the WP Path tracking to fully align with the target "runway". The length of the Approach phase depends on the cruise speed of the plane. This can be easily determined from a DVR by checking the distance with path tracking on, until the plane has stabilized its approach.

**Phase 3 - Glide Phase**
- the Approach phase ends when the Glide Waypoint is hit or the Glide Waypoint altitude is reached (in case of an undershoot)
- If a Sonar/Lidar is installed, Ground Altitude overrides Barometric/GPS altitude at any time
- at this point the throttle is set to 0% (or Idle if motor stop is disabled)
- althold is disabled
- pitch angle is set to `nav_fw_land_glide_pitch` (default 0°) and heading hold/course hold is activated (ideally set to the Approach heading that is planned, to compensate for any deviation during the approach phase)
- the plane will slow down in level glide and descent further while staying on track
- as soon as an altitude of `nav_fw_land_flare_alt` (default=1m) is hit (only if rangefinder is available to avoid stall if baro altitude measures too high) the pitch angle is set to a positive pitch by `nav_fw_land_flare_pitch` (10-15°) to initiate a flare and bleed of the last bit of speed till touchdown. Course hold is disabled to avoid any roll input. 
- **(to discuss)** If this is possible and a stabilized rudder + landing gear with steering is available, starting at the flare only heading should be controlled to allow a straight landing roll while keeping the wings level by the angle controller

[image]

### Current testing
With Logic conditions I have tested a manually created approach track like this and tested it in the HITL simulator under different wind conditions. Even with 20kph crosswinds + up to 8kph gusts, the wing was able to precisely touch down on the exact same line with not more than 2-3m deviation from a calm air landing. Field test with a LIDAR equipped plane is pending: 
https://youtu.be/icvKfaSUUdI

### Landing Spot Parameters for Safe Home and LAND Waypoint type
- Safe Home Points would need 3 extra parameters
1. Altitude above main sea level to define the exact ground level target altitude (Configurator and MWP can both load terrain altitude data, Relative altitude from Home can then calculated when arming)
2. Approach Heading 1
3. Approach Heading 2
- I suggest 2 heading values to allow 2 different approach directions that are not colinear to each other
- For both headings a positive number will allow INAV to land in that direction but also in the opposite direction if wind speed makes this necessary
- to make any heading exclusive (not allowing an opposing approach) the heading is put in with a negative number
- if only one heading is needed/possible, the user just puts the same number in both fields
- This allows up to 4 different approach directions if needed to make sure every wind condition can be covered

### SAFETY
- the whole landing procedure can be aborted by the user ANYTIME (except after the flare started) by pulling back on the pitch input
- the craft automatically disarms as soon as the speed falls below 2m/s to avoid any throttle incident
- If the wind direction does not allow a upwind landing because of limited available approach directions, the plane will just continue to loiter until either the pilots aborts the landing manually or the wind direction changes and allows a safe headwind or crosswind landing. This also compensates for initial wind speed or heading errors from very erratic flying that might take longer to correct than the initial 30s loiter
- a tailwind component limit should be set to still allow tailwind landings if the tailwind is below a safe threshold. For example a tailwind of 1-1.5m/s might still be safe or could even be a calculation error if it is extremely calm with no wind at all. Also a 90° crosswind could be interpreted as a very light tailwind component if the heading is not perfect. 
- (Any further suggestions?)

b14ckyy, 2022-12-12:
Here is an updated Video with some changes in the logic: https://youtu.be/FMsasMOk1b8
I have changed the altitude triggers to sticky triggers. This avoids the bounce in and out of the glide and flare part. 
Also a check is in place if AGL altitude is valid and it automatically switches between Altitude and AGL.

kasatka60, 2022-12-13:
Is it possible to implement a glide path on osd. For landings at night or with a foggy camera.

b14ckyy, 2022-12-13:
> Is it possible to implement a glide path on osd. For landings at night or with a foggy camera.

already implemented for 6.0. Glide Angle, Glide time and Glide distance.

kasatka60, 2022-12-13:
Sorry. Messed up the title. ILS
Like in pitlab
https://youtu.be/2wY4lGYUnoE

b14ckyy, 2022-12-13:
Not possible as we don't have the necessary graphical canvas OSD hardware needed for that. FrSky Pixel OSD had the capability but died. The other existing solutions either are also EOL or proprietary tech.

b14ckyy, 2022-12-17:
Added a markup explanation for the logic.

b14ckyy, 2022-12-18:
Real flight test successful. 4 Landings. All buttery smooth. Beside one deciation of approximately 9m to the side, the other 3 landings where all within a 2m corridor. https://www.youtube.com/watch?v=GCvHRhDJ0So

mmosca, 2022-12-18:
Nice job!

It would be great if INAV could calculate the landing parameters based on launch location/heading.

b14ckyy, 2022-12-18:
@mmosca that's the final idea. But not from launch point as we cannot know if the arm location is suitable for approach and land. It will be only for safehome or Land Waypoints in missions.

sensei-hacker, 2023-01-08:
@kasatka60  While we don't have the same graphical OSD, you CAN have an above/below glideslope indicator on the OSD.  I use one.

I have mine set so it shows "10" if I'm perfectly on glideslope. Higher or lower numbers indicate I'm below or above the glideslope. 

[
[image]
]

Then int he OSD page select to display gvar 0.

The numbers I chose is I show the glideslope for touch down 8 meters before the home point, and I have the target glideslope as 9:1.  That's the "8" and "90" in the image.

b14ckyy, 2023-02-20:
> > logic 14 1 11 45 0 1 0 -5 0
> > logic 20 1 19 45 0 1 0 -10 0
> 
> Hi, why the pitch angle you chose is negative 5 in logic 14 and negative 10 in logic 20, not positive 5 or positive 10? It looks higher pitch angle can help slower the speed...

Negative values mean pitch up, positive values pitch down.

cfitches, 2023-03-09:
Does this handle the fail safe scenario? i.e. RC link lost? In that if the wind isn't ideal and it runs out of approaches it can't just loiter forever? Wouldn't any landing approach be ok in this case even if the wind sucks?

b14ckyy, 2023-03-09:
> Does this handle the fail safe scenario? i.e. RC link lost? In that if the wind isn't ideal and it runs out of approaches it can't just loiter forever? Wouldn't any landing approach be ok in this case even if the wind sucks?

This is actually a good point. Usually in most locations, have a landing strip that can be approached from 2 directions is already enough to never land in a tail wind. 
Yes there might be some rare locations where only one direction of landing, or 2 directions at a <180° angle is possible, leaving a gap for a possible tailwind only landing. 
the Tailwind part threshold should also be configurable. 

Alternatively it could be programmed in a way that INAV just looks for the least amount of tailwind and does not reject landing at all. At least in a continous failsafe situation in case the RC system compeltely fails.

kasatka60, 2023-03-09:
How does inav determine the direction of a landing strip?

b14ckyy, 2023-03-09:
@kasatka60 that's all described in the first post.

closed 2024-03-20 by b14ckyy