iNavFlight/inav #8629: iFLIGHT BLITZ F722 Barometer and Magnetometer not recognised

bishko, 2022-12-12:
## Current Behavior
MODEL: Stock confighuration of iFLIGHT BOB57 (old Vista version, not O3 Air Unit version)
After flashing INAV 5.1.0 (latest stable) to iFLIGHT BLITZ F722, the barometer and magnetometer are not recognised.

## Steps to Reproduce
1. Flash INAV 5.1.0.
2. Load defaults for 3-7" quad.
3. Go to "Configuration" tab 
4. Change I2C speed from 400 to 800kHz.
5. Chnage Barometer from "None" to "DPS310"
6. Change Magentometer from "None" to "QMC5883".
7. Press "Save and reboot"

## Expected behavior
Both magnetometer and barometer should be recognised but are not. 

## Suggested solution(s)
Check the firmware target for iFLIGHT BLITZ F722. There seems to be some issue with how the I2C is configured.

## Additional context
- FC Board name and vendor: iFLIGHT BLITZ F722
- INAV version string: 5.1.0 (76f22b25)

b14ckyy, 2024-03-22:
If this is still n issue with 7.0 or 7.1 We will reopen. Close for now

closed 2024-03-22 by b14ckyy

b14ckyy, 2024-04-16:
@antichamber try the build from my post here: https://github.com/iNavFlight/inav/issues/9826#issuecomment-2052084512
you probably have a new version and I added the ICM gyro to the target for 7.1.1 patch. 

Baro is porobably a FC design issue. It should work if you connect the battery first and then USB. Seems like that baro error happens only when FC is powered by USB alone.