iNavFlight/inav #8636: Add Mavlink timesync protocoll

Consti10, 2022-12-19:
## Current Behavior
Being able to ping your FC via mavlink is a really usefully feature for testing / develpment. It looks like inav does not support timesync
(which replaced the ping protocoll, see [discussion on ardupilot](https://github.com/ArduPilot/ardupilot/issues/21311)

## Suggested Solution
Add timesync protocoll to inav mavlink, perhaps in a client-only version - this would allow any modern GCS like QGroundControll,
QOpenHD to ping the inav FC via mavlink (+ when done properly even calculate latencie(s) and more of the link)

## Who does this impact? Who is this for?
All users using inav with mavlink telemetry

b14ckyy, 2024-03-21:
Not sure if that is even possible as INAV does not support the full Mavlink Serial Protocol. Its only telemetry and RC over mavlink. The mavlink capability is very limited in general and afaik there are no plans to extend that.

b14ckyy, 2026-09-22:
Resolved in #11715

closed 2026-09-22 by b14ckyy