iNavFlight/inav #8642: SkyStars H7 HD (H743) - new target ?

alccrl, 2022-12-20:
I cannot configure in inav 5.1 or 6.0 beta flight controller SkyStars H7 HD.
It comes with betaflight installed (upgradable from github with Configurator) and ArduCopter as an option.

Is there any compatible target in Inav, at present ?
Is goig to be added , if not ?

Alfredo

alccrl, 2022-12-25:
Hi,
I am trying to define this new target in spare time, copying resources from betaflight config file for SKYSTARSH7Hd, but I have doubts about flash device settings.
This Fc seems to use w25q128fv chip that has no inav direct references (.h .c driver files).
They could be copied by betaflight but ... 
target compiles also without them, so I am expecting problems at runtime....
(stale #define directives)

Does anybody know if is it sufficient to copy/adapt betaflight equivalent files in order to make w25q128v flash device to work (modularity is enough tested), or is it necessary to review entire flash handler chain in the source code after adding a new flash device ?

Any idea ?

Alfredo

alccrl, 2023-01-01:
Further investigation suggests that this flash chip is supported.
Not yet tried the all thing, anyway....
it seems that arducopter implementation (features allocation) is not the same as betaflight, for this target, so this makes a bit confusion to me as how to proceed in Inav for Skystars H7 HD target addition.
As soon as I will (hopefully ...) succesfully test what I am doing, I will log results here.

Alfredo

alccrl, 2023-01-03:
First trial results:

MAG, BARO, OSD, RECEIVER, GPS OK.
GYRO/ACC not recognized (many modifications of target.c and target.h tested, looking at existing examples, ...no way ...
Pinout seems to be correct, but MPU is not clearly defined (BMI270 ?, MPU6000, both ?)
[CMakeLists.txt](https://github.com/iNavFlight/inav/files/10338600/CMakeLists.txt)
[config.c.txt](https://github.com/iNavFlight/inav/files/10338601/config.c.txt)
[SKST-SKYSTARSH7HD.config.txt](https://github.com/iNavFlight/inav/files/10338602/SKST-SKYSTARSH7HD.config.txt)
[target.c.txt](https://github.com/iNavFlight/inav/files/10338603/target.c.txt)
[target.h.txt](https://github.com/iNavFlight/inav/files/10338604/target.h.txt)

Is there anybody that can help fixing target.c and target.h files for gyro/acc to finally work ?
From H7 HD specs (SkyStars) it seems that there is a BMI270 ACC/GYRO but I do not understand if this system must be configured for DUAL_GYRO or something else, because betaflight config file shows two gyros and double pinout.

Alfredo

alccrl, 2023-01-07:
SUCCESFUL !!!,
prototype TARGET SKYSTARSH7HD now WORKS....

Looking at differences between BF and INAV, I found a compromise in configuring Gyro and Acc (just a cross-check between sources, not fully investigated all the procedure chains).

Here attached files target.c and target,h (final version).
[target.c.txt](https://github.com/iNavFlight/inav/files/10367214/target.c.txt)
[target.h.txt](https://github.com/iNavFlight/inav/files/10367215/target.h.txt)

I am not going now to pull gihub requests for this.
If anybody is interested about this target, write here (in this issue).

Alfredo

alccrl, 2023-01-15:
I succeded in getting ibus telemetry (https://github.com/OpenI6X/opentx/discussions/324) for gps with my SKYSTARSH7HD home made target,
gps, mag, vtx, external frsky osd (https://github.com/betaflight/betaflight/issues/12147), everything basic in order to fly and get rth or nav-pos hold seems to work, but quad does not take off...

Rear motors do not spin up properly,

flashing BF 4.3.2 this problem is not present, motors spin up correctly.
it could be a problem in TIM/DMA resource setup ? ...I copied resources from BF unified target.
I attach here the files if anybody is interested in confirming behaviour or for investigation.
[config.c.txt](https://github.com/iNavFlight/inav/files/10421105/config.c.txt)
[target.c.txt](https://github.com/iNavFlight/inav/files/10421106/target.c.txt)
[target.h.txt](https://github.com/iNavFlight/inav/files/10421107/target.h.txt)

alccrl, 2023-01-15:
Also flash chip (should be w25q128) (16MByte Blackbox) is marked as incompatible.

alccrl, 2023-01-15:
Note 1:
(if anybody is intersted in this port to inav from bf)

I am using this FC in a bit unusual way, because I am porting to inav/bf an old big quad (H809S GPS) that has brushed motors, changing the original custom flight controller with a SKYSTARSH7HD FC.

Since I did not find any documentation that declares this is not possibile, I am carrying on this port.
This old quad, to be honest, is very well made, very stable (11 inches props) and good mechanics, only rc link was weak factored, and adding fpv video and flysky 2..4 ghz system to it, should improve very much its features.
I succeded also in attaching an heavy "walkera" cheap 3axis gimbal, and footage looks very stable.

Note 2:
I am using 2s LiHv (3900 mah), not 3s lipo battery, Skystars Fc is declared for 2s, bottom stack for 3s (because of becs, but I am not using becs), because of mechanical constraints.
I investigated if 2s was reason for c1/c2 camera selection not working, but I tested also with 3s batteries, and I was not able  to use internal osd, ...either.

Alfredo

RedProps, 2023-01-15:
Hi I would like to use your target but I don't know how to load it into the inav configurator. I get a message in the firmware flasher sating the hex file is corrupt.

alccrl, 2023-01-16:
I did not upload the hex file, I uploaded the target folder sources.
I will try to upload the hex file into the next 24h, hopefully.

Alfredo

alccrl, 2023-01-16:
If You are in a hurry and You are able to compile, just create a target dir SKYSTARSH7HD among other targets, and build.
At the moment I have no hex on my phone.
Files must be taken off .txt extension before build.

Alfredo

alccrl, 2023-01-16:
Addendum (sorry):..
In order to build it is necessary to create a makefile.in the target dir.
Copy another target makefile and change the target name inside.

I will copy hex and put a few notes later ( not everything works).

Alfredo

alccrl, 2023-01-16:
Hi,

[inav_6.0.0_SKYSTARSH7HD-alccrl-0.1.0.zip](https://github.com/iNavFlight/inav/files/10427787/inav_6.0.0_SKYSTARSH7HD-alccrl-0.1.0.zip)
Here You find hex file is in zip format (must be unzipped first).

Notes:
with this sw I have a problem with motor spin.
Rear motors do not spin well in my configuration.
At presnt I don't know if it is a configuration issue or sw issue.
Since You much probably will use **DSHOT*** protocol, let me know how sw behaves with You.
I am using BRUSHED protocol and not standard mixer motor order (I used mixer wizard).
I have yet to investigate better what is going on, rear motors do not have enough speed to take off.
It shoudl not an electrical problem, because under Betaflight 4.3.2 they work.
Use 6.0 configurator, this is inav 6.0 firmware.

Alfredo

RedProps, 2023-01-16:
HI Thank you so much I will be flying a tricopter so I will provide feedback

alccrl, 2023-01-26:
A fast update:
Using BF 4.3.2 I discovered that rear motor mosfets for my brushed drone were damaged, probably after some test with 3s, 4s batteries instead of original 2s batteries.
Infact at switch on of quad, high voltage peaks or something similar might have partly burnt mosfets.
Now I got problems in PID tuning, my big brushed  quad is far from been tuned as a brushless 5" quad ...
After changing broken motors, quad is not stable after take off, oscillations increase, defaul pid system is not stable.
Conclusion ...
is that I did not yet fully test my inav 6.0 beta skystarsh7d target.
Other feedbacks are welcome.

Alfredo

vishal2248, 2023-02-01:
I have same fc skystar H743 HD dual gyro

Your work proper inav in fc ?

alccrl, 2023-02-01:
As I wrote above, 
it seems to work, but my motors are brushed, and I have yet to find correct PIDs to fly.
Everything else, except flash for blackbox and camera input, works.
Ther is something abput camera selection that I have still to understand. Usong an external frsky osd, goggles get correct osd info and camera input.

Alfredo

alccrl, 2023-02-01:
This target is not yet in inav official stream.

vishal2248, 2023-02-01:
And betaflight camera black screen only work osd
Ardupilot work fine camera

alccrl, 2023-02-02:
I tried also imverting pinio1 and pinio2 flags in source code to pull-high pc5 pc2 like arducopter, but it does not work anyway..

Alfredo

alccrl, 2023-02-02:
Just an idea ...

I cant exclude existence of more Skystars H7 HD stocks that differ from original used for Arducopter. This is not probable, but not impossible, because labels on FC do not correspond exactly to (poor) documentation about.

Alfredo

alccrl, 2023-02-04:
HI,

I attach here the final version of my custom build for SkystarsH7HD 5.1 target that currenty is not included in Inav.
May be useful for some tester, or someone owning this fc.

[inav_5.1.0_SKYSTARSH7HD.zip](https://github.com/iNavFlight/inav/files/10609599/inav_5.1.0_SKYSTARSH7HD.zip)
[SKYSTARSH7HD.zip](https://github.com/iNavFlight/inav/files/10609600/SKYSTARSH7HD.zip)

- One file contains target directory zipped with sources, 
- the other file contains the hex executable to be flashed using configurator in "local" mode.

I changed (last trial) an AF (Alternate Function) definition for flash chip that was different from betaflight unified target, but blackbox is still recognized as not supported (w25q128 chip) by configurator (may be configurator issue (?) - I had a fast look at configuratot sources but it seems that compatibility is calculated inside FC, non configurator ...).

Summary:
I did not yet fly this target (see bottom notes), but
- ports
- gps
- ibus rx
- ibus tm
- vtx irc tramp
- frsky osd
- mag
- barometer
- accel/gyro
work.
Flash BlackBox and Camera cam1 cam2 selection (user 1 user2 modes) do not work.

Why did I not yet fly ?
Because for this 9 inch prop "build" I am using BRUSHED protocol/motors and about this I encountered documentation, hw and sw problems that now should be solved.
Moreover I tried for comparison to use this target with BF 4.3.2, BF 4.4, Inav 6.0 beta and Arducopter, but I found a lot of problems, no way to come out of them (PID and preset included that seem all to behave same way, or the quad flies away just armed or more or less oscillates and falls down, no middle choice)  last was (as an example) that BF 4.4 does not recognize external frsky OSD and new BF 4.4 cloud build is a mess. 
So ... I am confident that using this target with common 5 inch fpv brushless quads or fixed wing,  should be a more enjoying and straightforward task.

Hope this helps,

Alfredo

alccrl, 2023-02-09:
Tiny update:
1)I discovered my skystars h7 hd target to be single gyro (label on package ...:-) but on skystars site pictures show dual gyro.... pinout different ? Ardupilot uses dual gyro ? Who knows  ...
2)I am planning to do a fly test in acro mode, because anchored tests show all motors to work properly with , let's say, fast response to sticks.
After 10 min speed test and powercycle, dont know why,  motors arm again all 4 at same speed. Rusty motors (not used very much ...) ?. So I am confident this "thing" will fly sooner or later ...

Alfredo

alccrl, 2023-02-11:
IT FLIES...I tested it, also poshold works.

End of the story ...
Nobody seems interested in this target skystarsh7hd,, so I stop writing.
I will work on flash for blackbox compatibility (I di not find a clear statemente about w25q128 compatibility or uncompatibility), but just for personal curiosity.

Alfredo

closed 2023-02-11 by alccrl

alccrl, 2023-02-12:
Post "closure" :-) update:

also flash for blackbox works, after adding 0xEF6018 as Winbond flash ID
in file flash_m25p16.c 
(...)
#define JEDEC_ID_WINBOND_W25Q128_3     0xEF6018
(...)
        case JEDEC_ID_WINBOND_W25Q128:
        case JEDEC_ID_WINBOND_W25Q128_2:
        case JEDEC_ID_WINBOND_W25Q128_3:
(...)

One problem only remains for this experiment/target skystars h7 hd:

camera video blank screen toward vtx video output (input video does not pass through).
I guesss this to be a pinout misunderstanding (documentation is very poor about this, may be working Arducopter uses dual gyro version that might have different pinout, who knows ...) on camera selection GPIO.

Alfredo

alccrl, 2023-02-22:
I investigated the hw.

B7AJ switch might be responsible for failiure of video in passthroigh to osd chip.
Could be a soldering problem.
PC2 pin is correctly addressed.to fc from the switch.
I detached the switch and connected directly video in cam1 to the capacitor of video in in osd chip.

Now it works.

Alfredo