iNavFlight/inav #8647: Probleme Raper extrem Foxeer Inav

Fabwize, 2022-12-22:
how can we enter the VTX tables of the reaper extrem foxeer dan inav 5.1 and inav 6

when I look at the frequencies and power in the osd, I only see 0 in power and ..../0 in frequency as the value that appears.

thanks.

- FC Board name and vendor: F411WTE MATEK
- INAV version string: 5.1

MrD-RC, 2022-12-22:
There are no VTx tables in INAV.

Fabwize, 2022-12-22:
> Il n'y a pas de tables VTx dans INAV.

is there a solution to make the VTX reaper extrem work with the smart audio correctly ?

MrD-RC, 2022-12-27:
@Fabwize I don't have that VTx. Could you try this to see if the power works.

[inav_6.0.0_MATEKF411TE.zip](https://github.com/iNavFlight/inav/files/10308011/inav_6.0.0_MATEKF411TE.zip)

Fabwize, 2022-12-27:
@MrD-RC 
hello, I have already tried, when the frequency and power are displayed on the OSD, it displays ../0
the power remains blocked in 25mw while I ask for tramp power 5=600mw=2.5w Foxeer Reaper extrem

Fabwize, 2022-12-27:
@MrD-RC thanks, i will try quick

Fabwize, 2022-12-27:
@MrD-RC  this file dont work in CLI
where i ave to put?

MrD-RC, 2022-12-27:
> @MrD-RC  this file dont work in CLI
> 
> where i ave to put?

It's firmware. You need to flash it to the flight controller. Make sure to make a 'diff all' of your current setup. You will need to use the 6.0 FP2 version of Configurator too. I this works, I can submit the change to be in INAV 6.0.

closed 2022-12-27 by DzikuVx

Fabwize, 2022-12-27:
@MrD-RC I just flashed, it does not work, there is no more compass, GPS, pitito, there is nothing in blue at the top of the page and the frequency and power channels do not change either.

MrD-RC, 2022-12-27:
That is setup. The things that changed had nothing to do with the compass etc.

Fabwize, 2022-12-27:
@MrD-RC 
[image]
[image]

Fabwize, 2022-12-28:
IF is possible, i need target for F411TWE and H743Wlite . is for F411TE fror nanodrak, H743Wlite for caipirina V2.
thanks.

MrD-RC, 2022-12-28:
This firmware was just to test, not to fly with.

Fabwize, 2022-12-28:
@MrD-RC thanks i anderstand. but no irc tramp connexion no work.
i cant change fréquency , band and power level

Fabwize, 2022-12-28:
I welded in serial 2 on the FC

[image]

MrD-RC, 2022-12-29:
[inav_6.0.0_MATEKF411TE.hex.zip](https://github.com/iNavFlight/inav/files/10321894/inav_6.0.0_MATEKF411TE.hex.zip)

I did notice that there was a discrepancy in the power names. So this may have caused an issue. If not, then it is more than just the power table that is the issue.

Fabwize, 2022-12-29:
@MrD-RC  i juste try, exacly the same. the vtx dont change fréquancy and chanel, only the osd change.
thanks

Fabwize, 2023-01-01:
@MrD-RC Happy new year, 
do you whant i send to you the VTX?
i dont want no money, i send you free. 
just if you can re send to me, when you finish work? if you cant , is not too bad.

Fabwize, 2023-01-03:
@MrD-RC Hello, i progress
 mabe is beacose the F411twe dont ave serial port.
ST2 is soft serial. ans smart audio good work on soft serial, but irc tramp dont work.

i ave H743Wlite. 
can you give to me a target to try with dem?
and tel me who to soldering .

[image]

MrD-RC, 2023-01-03:
I've used IRC Tramp on SoftSerial with a Matek VTx and it works fine. Unfortunately I have no time to look at this at the moment. I am away with work.

Fabwize, 2023-01-03:
Thanks for the answer, i anderstand, no time everytime.
soft serial and irc tramp is a bad combo. sometime work sometime dont work, i try all ready with the originaly VTX Tramp.
dis is beacose cant change baude or nothing , this soft serial is a emulation da uart.
i will try on H743Wlite and comeback with you.
thanks again for your time and work.