iNavFlight/inav #8648: INAV 6.0.0 FP1

Orvo300, 2022-12-22:
## Current Behavior

In the OSD the IMU temperature jumps from 17°C to -28°C and only shows these two values.
Observed same behaviour on 3 FCs after flashing to INAV 6.0.0 FP1 (two F411-WSE and one F722-WING).

## Steps to Reproduce

1. flash FC with INAV 6.0.0 FP1
2. enable IMU temperature on OSD

## Expected behavior

show IMU temperature readings, just like in INAV 5 and previous.

## Suggested solution(s)

i have no idea

## Additional context

https://pastebin.com/zGqx06tK
---

- FC Board name and vendor: Matek F411-WSE and F722-WING
- INAV version string: INAV/MATEKF411SE 6.0.0 Oct 20 2022 / 15:30:00 (d1f6f19a)

MrD-RC, 2022-12-22:
This issue has already been fixed.

kasatka60, 2022-12-22:
https://github.com/iNavFlight/inav/issues/8538

closed 2022-12-27 by DzikuVx