iNavFlight/inav #8650: Angle Flight Mode for Boats and Rovers

DaveBow12, 2022-12-27:
## Current Behavior

## Desired Behavior

## Suggested Solution

## Who does this impact? Who is this for?

## Additional context

DzikuVx, 2022-12-27:
But what would Angle Domin those cases?

stronnag, 2022-12-27:
In any case, the hydrodynamics involved are way beyond the capabilities of INAV. Maritime auto-pilots are a very different domain, sailing APs even more so.

DaveBow12, 2022-12-27:
Angle can easily adjust hydro foils just as it sdjusts the ailerons and elevator of fixed wing planes to keep the boat level.
Yes hydro dynamics are different and more powerful servos may be required.  Others have already been doing this by programming there boat as an airplane and they have made it work.  There are also boats that use air wings, just like race car wings, to keep boats level.  I expect this is an easy change so why not make Angle availabe in the boat and rover configuration?

closed 2023-04-18 by DzikuVx