iNavFlight/inav #8667: BF43COMPAT mode for DJI O3 air units, OSD freezes afer 1-2 minutes once airborne

OlivierC-FR, 2023-01-01:
INAV 6.0 master branch, built 21-12/2022
FC : ATOMRCF405NAVI

While it seem to work fine on the bench, the OSD on the O3 stops updating after a delay of about 1 to 2 minutes once the plane has been launched. All values and texts freezes and stay there forever until the plane is landed and rebooted.

The O3 itself it not crashing nor overheating, it continues to record the onboard HD and stream the FPV feed etc
Happens with 2 different O3 units and 2 different way to power supply the O3, and 6 times over 6 flights.
Does not happen with Arduplane 4.4-dev, same plane and FC, there the OSD works as expected.

diff : https://www.dropbox.com/s/0tjp9qn0ekw3wbb/INAV_6.0.0_cli_20221221_104224.txt?dl=0

kingdraki, 2023-01-02:
I can confirm this with # INAV/MATEKF722SE 6.0.0 Dec 25 2022 / 07:36:56 (7f38d107)

If i do switch OSD Layout after freezing, it is working again for a minute or two until it freezes again.

Sil20, 2023-01-03:
I an add a symptome : using the CLI from the configuration make the OSD disappear. So this could be a way to trigger the issue quickely

mmosca, 2023-01-04:
I will try to reproduce the issue once I am home (in a couple of days) @OlivierC-FR just for completeness, can you refresh my memory if you googles 2 or v2?

OlivierC-FR, 2023-01-04:
> I will try to reproduce the issue once I am home (in a couple of days) @OlivierC-FR just for completeness, can you refresh my memory if you googles 2 or v2?

V2, not rooted, last DJI firmware

OlivierC-FR, 2023-01-07:
Can confirm that changing OSD page is a valid workaround, it unfreezes the OSD display for a variable duration, I was expecting something like 60 or 120 seconds maybe but no, it's varying between approximately 2 to 4 minutes.

mmosca, 2023-01-10:
Just an update,  I did some quick bench testing during lunch time today but I have not been able to reproduce it yet.
I wonder if it is related signal quality between the o3 and googles. Maybe as you fly you get some lost packets between o3 and googles that cause the osd update to freeze.

From memory, I think there were issues in wtf-os where the inav osd would get corrupted and this would "reset" the osd to a working state.

From my understanding wtf-os fixed the reliability issue by having a vtx "frame buffer" and using compression to transmit the full frame over the air.

@OlivierC-FR your osd setup seems to have a single page. Can you try setting up multiple pages and use a switch to change pages or turn the osd on/off.

You can also try to simplify the OSD display to include less data. The theory is that the more fields you have, the more data has to be transferred between o3 and googles for the OSD to work well.

My test was not with the lastest version from master, and I did not have the exact osd setup as you.
I will do some further investigation on my side as well to try and reproduce it.  I will update inav to the latest master and check the dji firmware on my googles and air units in case I have different firmware versions from you.

mmosca, 2023-01-10:
@OlivierC-FR missed your last comment before I replied.

Can you check if the PR #8654 for walksnail by @MrD-RC fixes the issue? It includes logic to send a full osd frame peridicaly.

You may need to make sure osd_msp_displayport_fullframe_interval is > 0.

Since changing pages fixed the issue for you, it is probably related to signal integrity and I suspect the P frame will ensure it recovers automatically.

MrD-RC, 2023-01-10:
> You may need to make sure osd_msp_displayport_fullframe_interval is > 0.
> 

Don't set it to 0, unless you want your OSD locked up. The default of 5 seems fine. Setting it to 1 works, but is likely overkill.

OlivierC-FR, 2023-01-11:
> 

I think I made the HEX with the lastest master at the time, including this PR 8654, but I'm not 100% certain, so I will rebuild a fresh Hex and reflash and will confirm. Might take few days, horrible weather here. And yes I have a switch to change OSD page, 3 positions, it's set to default page / alternate osd 1 / OSD off. About signal between O3 and goggles, the OSD lock happens even at short range, with the O3 in auto/50mbs mode, ranges where the video signal is always rock solid. It's annoying the V2 cannot record the OSD layer, when flying it's not easy to know how long it takes to freeze etc...

mmosca, 2023-01-11:
@OlivierC-FR I will try harder to reproduce it. I have the same setup as you, other than the FC. I just updated the firmware on the O3 to the new V01.02.0000 and installed a recent master build.

mmosca, 2023-01-11:
I finally managed to reproduce the issue.

Good news, I know what command needs to be sent to the air recover the updates.

The bad news is that the avatar changes do not help the updates to restart and it is likely an issue on the DJI side.

Sending a clear screen subcommand will resume the update, but it also causes the osd to flicker.

I will investigate the best way to send a clear screen with minimum flicker (and make it an optional behavior).

mmosca, 2023-01-11:
@OlivierC-FR can you try this firmware?

[inav_6.0.0_ATOMRCF405NAVI.zip](https://github.com/iNavFlight/inav/files/10396121/inav_6.0.0_ATOMRCF405NAVI.zip)

It won't stop the freezing, but it should recover automatically.

OlivierC-FR, 2023-01-12:
> @OlivierC-FR can you try this firmware?
> 
> [inav_6.0.0_ATOMRCF405NAVI.zip](https://github.com/iNavFlight/inav/files/10396121/inav_6.0.0_ATOMRCF405NAVI.zip)
> 
> It won't stop the freezing, but it should recover automatically.
> 
> You probably should use the configurator from @MrD-RC, but loading a diff should work even with the configurator version from main.
> 
> His configurator is here: https://www.dropbox.com/s/glqvnr9ps2huq5x/INAV-Configurator_win64_6.0.0.zip?dl=0

ok thanks, will try (airborne) this week-end

Sil20, 2023-01-12:
> Just an update, I did some quick bench testing during lunch time today but I have not been able to reproduce it yet. I wonder if it is related signal quality between the o3 and googles. Maybe as you fly you get some lost packets between o3 and googles that cause the osd update to freeze.
> 
> From memory, I think there were issues in wtf-os where the inav osd would get corrupted and changing pages would "reset" the osd to a working state.
> 
> From my understanding wtf-os fixed the reliability issue by having a vtx "frame buffer" and using compression to transmit the full frame over the air.
> 
> @OlivierC-FR your osd setup seems to have a single page. Can you try setting up multiple pages and use a switch to change pages or turn the osd on/off.
> 
> You can also try to simplify the OSD display to include less data. The theory is that the more fields you have, the more data has to be transferred between o3 and googles for the OSD to work well.
> 
> My test was not with the lastest version from master, and I did not have the exact osd setup as you. I will do some further investigation on my side as well to try and reproduce it. I will update inav to the latest master and check the dji firmware on my googles and air units in case I have different firmware versions from you.

the issue on WTFOS was not really that the buffer got corrupted but was not transmitted at all. And since INav optimize the data transmission, ony deltas are sent to the OSD  chipset or MSP Display port. The behaviour in such case was a horizon "_" char staing as long as another char was send to be displayed.

OlivierC-FR, 2023-01-13:
> @OlivierC-FR can you try this firmware?
> 
> [inav_6.0.0_ATOMRCF405NAVI.zip](https://github.com/iNavFlight/inav/files/10396121/inav_6.0.0_ATOMRCF405NAVI.zip)
> 
> It won't stop the freezing, but it should recover automatically.

Works !
Made a 20kms long flight, about 20 minutes, distance up to 4kms, and no freeze, did not have to wiggle the OSD switch.

Sil20, 2023-01-14:
Would you have a binary for inav_6.0.0_? I don't think it is already in the current Dev Build

mmosca, 2023-01-14:
> Would you have a binary for inav_6.0.0_? I don't think it is already in the current Dev Build

You can get binaries from the artifacts in the pull request linked to this issue.

https://github.com/iNavFlight/inav/actions/runs/3917662388#artifacts

closed 2023-01-14 by MrD-RC

Sil20, 2023-01-14:
> > Would you have a binary for inav_6.0.0_? I don't think it is already in the current Dev Build
> 
> You can get binaries from the artifacts in the pull request linked to this issue.
> 
> https://github.com/iNavFlight/inav/actions/runs/3917662388#artifacts

perfect thank you.
I just tested  on my setup and I get the following system message : 
invalid setting 
OSD_MSP_DISPLAYPORT_FULLFRAM

Which I think is truncation for the "osd_msp_displayport_fullframe_interval " parameter. which value is default : 10
I did not have this with the previous build  "inav_6.0.0_MATEKF405SE_ci-20230102" and I just past the diff after

I will look around what could be the issue but if it rings a bell on your side please advice.

mmosca, 2023-01-14:
The fix is now in the main branch. I think the value has changed from seconds to 1/10 seconds.

Maybe you need to update to the latest configurator from the main branch as well, or use the cli to set the value, but the default is 1s, so it should be fine.

If you are still having issues with the latest main version, you can always report it as another issue :)

MrD-RC, 2023-01-14:
There was an issue with the PR at one point, which has now been fixed and the PR merged. 

Just to clarify, use the default value of 10, which is 1 second.

Sil20, 2023-01-15:
> which has now been fixed and the PR merged.
> 
> Just to clarify, use the default value of 10, which is 1 second.
Thank you.
I am not familiar enough to find a build for the main though I looked a bit is it available somewhere or should I build my own?

bakwc, 2023-01-22:
Have the same issue on a fresh master build. OSD freezed after 15 minutes flight time and stayed with same values until landed.

mmosca, 2023-01-23:
@bakwc 

> Have the same issue on a fresh master build. OSD freezed after 15 minutes flight time and stayed with same values until landed.

Have you tried switching osd pages to see if it still helps?

What goggles were you using?

Do you have a dvr video with osd overlay?

bakwc, 2023-02-18:
Looks like the issue is in goggles v2. May be it's because of wtf hack installed. All OSD freezes (including bitrate and other elements) until googles restart.

RobinatiNav, 2023-03-02:
This is not really to the topic but I have DJI O3 air unit and goggles 2 and with this setup iNav OSD works with no problem.

https://youtu.be/x8POPAefDPE

Swainic, 2023-04-05:
I have a Matek F411 TE on INAV 6.0 and have the OSD for the 03 Air Unit on one of the 2 serial ports (due to the lack of UARTS). Port is set to MSP Displayport and OSD set to BF43COMPAT. I have been scratching my head for a couple of days trying to get the OSD on screen in my DJI V2 Goggles. After reading the above posts on the OSD screen options I tried experimenting with that area. I have my screen options on S1 - Taranis pot. I found that the default screen comes on at a very small specific range of the pot movement (about 11 o-clock setting). Then I tried setting up option screen 3 as a test and that range works OK within the '3' range of pot movement but there is some flickering of a few icons on the screen, less stable than the default area. So if a switch is used for the OSD options I can see it not working at all on the default screen. I have the stock INAV firmware at the moment, and this is all on the bench so far, not set up in the plane flying yet. I'll play around a bit more and perhaps try the firmware above to see if that changes anything. But the main thing is that I now have the OSD working on the 03 air unit, so thanks for the contributions above, they helped me a lot.

mmosca, 2023-04-05:
Can you share the screenshot of your ports tab?Softserial is not reliable/fast enough for o3.

Swainic, 2023-04-05:
Here is a screenshot of the ports page. I'm pretty new to this, maybe I can put the GPS on a serial? I have smartport from the receiver on the Serial 1 but i'm not sure that is doing anything? I have another plane running Caddx on the smartport of the same Matek F411 and it works just fine. 
[image]

mmosca, 2023-04-05:
The o3 and the rx 100% need a uart. Putting the gps on softserial is risky and you will probably see random problems as you saw if the o3 air unit. Softserial would work for slower protocols, like smart audio to control an analog vtx or other non critical features. Unfortunately, the f411 flight controllers are starting to show its limitations, since digital vtx are becoming more and more common.

bakwc, 2023-04-05:
O3 works OK on softserial (on Matek f411wte). GPS and RX on real UART is more important.

Swainic, 2023-04-05:
Thanks, I'll give it a try and see if it works

PR #8654 (2023-01-14): Update MSPDisplayPort and add Walksnail Avatar settings
- Add support for the Avatar OSD settings
- Add support for centring elements on OSDs with odd number of columns
- Add option to send P frame at defined intervals
- Rename MSP defines to match [API](https://github.com/betaflight/betaflight/blob/master/docs/API/DisplayPort.md), and remove duplicates
- Send attribute bits instead of Boolean for `MSP_DP_WRITE_STRING`, as per spec

Requires Configurator https://github.com/iNavFlight/inav-configurator/pull/1683

Fixes #8667