iNavFlight/inav #8669: icons for configurator do not work in master branch

fermiums22, 2023-01-02:
## Current Behavior

[image]

## Steps to Reproduce

1. compiled master
2. compiled branch tagged 6.0.0-FP1

## Expected behavior

Icon display. IMU works without icons and also has lags in 3D display.

## Suggested solution(s)

rollback to branch tagged 6.0.0-FP1
## Additional context

---

    # version
    # INAV/IFLIGHT_SUCCEX_E_F7 6.0.0 Jan  2 2023 / 09:53:58 (5937ba71)
    # GCC-10.2.1 20201103 (release)
- FC Board name and vendor:
https://www.banggood.com/30_5x30_5mm-iFlight-SucceX-E-Stack-F7-V1_4-2-6S-Flight-Controller-5V-2_5A-BEC-45A-V2-BLHeli_S-Dshot600-4-In-1-Brushless-ESC-compatible-DJI-Air-Unit-for-RC-Drone-FPV-Racing-p-1800168.html?cur_warehouse=CN
- INAV version string:
(5937ba71)

fermiums22, 2023-01-02:
branch 6.0.0-FP1 add IFLIGHT_SUCCEX_E_F7 https://github.com/fermiums22/inav

fermiums22, 2023-01-02:
Are there any critical fixes in the master? I can safely fly on old version?

MrD-RC, 2023-01-02:
It is known that the icons do not light in FP2 configurator when using firmware compiled from the master branch. You should use configurator from it's master branch too. 

Master is obviously in development. So all flights should be considered testing and are at your own risk.

closed 2023-01-02 by MrD-RC