iNavFlight/inav #8673: Failsafe issue when extra arming safety off

macgyverek, 2023-01-03:
## Current Behavior
I found a problem with Failsafe when nav_extra_arming_safety == OFF and RTH is enabled. Without GPS fix drone making fly away. It not safe flyting at home, buildings etc.. Its a dangerous !!! The same issue was on BF and was fixed. 

## Steps to Reproduce
1. nav_extra_arming_safety == OFF
2. Enable RTH Failsafe
3. ARM quad
4. Power Off Radio
5. Failsafe making fly away 
6. Power On Radio
7. Disarm quad
8. Motors still working - only power cut can stop the motors. 

## Expected behavior
Cut off motors immediately 

## Suggested solution(s)
If nav_extra_arming_safety == OFF and arm quad without gps fix/home point inav should drop or land when RX lost. 

- FC Board name and vendor: Mamba F7
- INAV version string: 5.1

MrD-RC, 2023-01-03:
OFF has been removed from nav_extra_arming_safety in 6.0 for exactly this reason. Switching that off and arming without a GPS lock is at your own risk.

macgyverek, 2023-01-03:
So flying without GPS fix will be not possible ?? What about flying in hall, buildings ?? Do you have a new solution for cinematic drones or inav is opening only for flying wings / plain etc. ??

MrD-RC, 2023-01-03:
@macgyverek For flying indoors you can use allow_bypass instead of off. Yaw right when you arm then it will be as if you have turned nav_extra_arming_safety off, until you power cycle the battery. 

I don't see how this would effect a cinematic drone at all?

macgyverek, 2023-01-03:
I will check what will be when I use allow_bypass and rxlost (failsafe). 

About cinematic drone: sometimes I uses drone making a shoots flying outdoor and indoor. So better is flying without GPS/Home Point. After lost link drone should land or disarm motors/drop. In my opinion inav should automatically disable RTH when armed and no fix. Home point should not be set. User should make a simple decision when start/arm: start without GPS or wait for GPS fix and flight. Check DJI drones: they have that options and working good.

macgyverek, 2023-01-03:
I tested  allow_bypas. The same issue: drone making fly away without GPS signals after rxlost. Hmmm: where is flying ?? To China  or to Africa ??

breadoven, 2023-01-03:
> I tested allow_bypas. The same issue: drone making fly away without GPS signals after rxlost. Hmmm: where is flying ?? To China or to Africa ??

The Drone isn't flying anywhere it's merely drifting out of control. RTH won't initialise without a valid GPS fix so it's not active during Failsafe. In this case Failsafe RTH should default to Failsafe Landing automatically. Does it not appear to do this ?

breadoven, 2023-01-03:
Just tested this and it worked as expected. With RTH Failsafe set with no GPS fix it defaulted to an Emergency Landing, Landed and disarmed. This was for a fixed wing but it should work the same from a multirotor.

If you don't have a Baro available then the landing will just use `failsafe_throttle `to control the descent rate. Has this been set to a different value from the default of 1000  (effectively Drop on a MR) ? With a Baro it should descend at the rate set by `nav_emerg_landing_speed`.

macgyverek, 2023-01-03:
Yes I have a baro. 

I tested like this:
- Drone armed without gps fix (motors working at 10%)
- Drone are on the ground (not fly)
- Power off radio
- Motors are spinning up after failsafe (a hold it by hand), sometimes are only 2 motors are spinning (in normal flight will be flip). I feel on hand that is no a landing mode. Throttle had very high values. 

The same behaviour is with and without propellers.
I dont tested this on real flight at home, because can be fly away.

macgyverek, 2023-01-03:
You too waste may time. Happy Flying...I returned to BF. Bye ;) Sorry for this task.

lida2003, 2023-07-28:
> Yes I have a baro.
> 
> I tested like this:
> 
>     * Drone armed without gps fix (motors working at 10%)
> 
>     * Drone are on the ground (not fly)
> 
>     * Power off radio
> 
>     * Motors are spinning up after failsafe (I hold it by hand on the ground), sometimes are only 2 motors are spinning (in normal flight will be flip). I feel on hand that is no a landing mode. Throttle had very high values.
> 

Ah,,, interesting test. GPS lost + RC link fail triggered motor spin issue (sometimes 2 motors), it's kind of [ Emergency landing tumbled crash? -- inav 6.1.1 #9184 ](https://github.com/iNavFlight/inav/issues/9184), what I have met flipped crash.

> 
> The same behaviour is with and without propellers. I dont tested this on real flight at home, because can be fly away.
> 
> failsafe_delay = 5 failsafe_recovery_delay = 5 failsafe_off_delay = 200 failsafe_throttle_low_delay = 0 failsafe_procedure = RTH failsafe_stick_threshold = 50 failsafe_fw_roll_angle = -200 failsafe_fw_pitch_angle = 100 failsafe_fw_yaw_rate = -45 failsafe_min_distance = 0 failsafe_min_distance_procedure = DROP failsafe_mission = ON failsafe_throttle = 1000