iNavFlight/inav #8675: Add ACRO mode in INAV Programming Framework FLIGHT_MODE

sdellava, 2023-01-03:
## Current Behavior
The ACRO mode is missing in the modes list in the Programming Framework.

## Desired Behavior
Add ACRO mode in the Programming Framework FLIGHT_MODE table (and related sw)

## Who does this impact? Who is this for?
Every INAV users

## Additional context
There are some special situations where a programming logic is very useful but it is dangerous to apply if the aircraft is not in ACRO mode. In example it could be useful to automatically change the PID profile in specific fligth condition (altitude, speed, etc...) but only in ACRO mode. But ACRO mode is non one of the modes choosebale in the Programming framwork to build a logic condition.

breadoven, 2023-01-03:
ACRO mode isn't actually a mode in it's own right it's simply the "mode" that applies when no other flight mode is active. So if you want to be in ACRO mode you deselect all other flight modes.

sdellava, 2023-01-04:
It seems to me a very easy to add virtual state, that otherwise requiers 9+ logical condition to be evaluated.

MrD-RC, 2023-01-04:
I disagree that Acro isn't a mode. It's just that is doesn't have a specific switch. In that way, it's like the definitive flight mode. 

I think a LC for the acro mode is a fair enough request.

breadoven, 2023-01-04:
Agree that an LC for what is effectively "no flight mode" makes sense given the only other way of selecting it would appear to be by negating the other modes using numerous logic conditions.

Also, why is there no WP mode included in the LC flight modes ?

MrD-RC, 2023-01-04:
No idea. That should also be an option. There is **is WP** which could be the same thing, just in a different place?

PR #8747 (2023-01-29): Update Flight Mode operand in programming framework
- Add `ACRO` to `FLIGHT_MODE` operand
- Add `WAYPOINT_MISSION` to `FLIGHT_MODE` operand
- Update functionality of `COURSE_HOLD` as currently it will still be active if `CRUISE` is active
- Update Programming Framework.md document

Fixes #8675 

Requires Configurator https://github.com/iNavFlight/inav-configurator/pull/1705
closed 2023-01-29 by DzikuVx