iNavFlight/inav #8691: ExpressLRS Wifi connection to INAV

Lethal-Paw, 2023-01-08:
## Current Behavior
 Currently one is only able to access the INAV GUI via USB cable or a Bluetooth device. When trying to Connect to the Flight Controller via the ELRS WiFi Connection it fails to open a serial Port even When the MSP Connection for the RX was enabled in the Ports Tab.

## Desired Behavior
 I would wish to connect to the FC via the the ELRS Wifi Passthrough like Betaflight already established. JB's Video Minute 4:00 (https://m.youtube.com/watch?v=cyILQqxPpZI)

## Suggested Solution
 I am a lot but no software geek, however I belive since the ELRS Devs unlocked this feature for all their Recievers and it was implemented in the Betaflight Firmware right away there seems to be only a little change in code in order to Unlock the MSP Port for the CRSF Portocol of the ELRS recievers

## Who does this impact? Who is this for?
 It impacts everyone using ELRS Recievers who want to be able to connect to their FC wirelessly.

## Additional context
 I hope I did everything right with this post and it doesnt cause too much inconvinience. You INAV Devs are really doing an awesome Job! Thank you so much.

MrD-RC, 2023-01-08:
Duplicate of https://github.com/iNavFlight/inav-configurator/issues/1616

MrD-RC, 2023-01-08:
@Lethal-Paw This is an excellent example of a feature request. The only problem was it was added to the firmware repository, where this is more a configurator feature request. But other than that, it was good 👍🏻

closed 2023-01-08 by MrD-RC

Lethal-Paw, 2023-01-09:
Hi MrD, thanks for the feedback. Would you recommend to make this request again in the configurator section or do you think it will be implemented anyways?

MrD-RC, 2023-01-09:
@Lethal-Paw The request is already in configurator. See the duplicate request comment before I closed the issue.

Lethal-Paw, 2023-01-09:
Ah yes now I see it. And since the request is still open it seems to be a "work in progress".

RomanLut, 2023-01-14:
> @Lethal-Paw This is an excellent example of a feature request. The only problem was it was added to the firmware repository, where this is more a configurator feature request. But other than that, it was good 👍🏻

Actually this is firmware problem. There are no changes required in configurator.

ELRS is encapsulating MSP messages which are sent to tcp port by configurator into CRSF packets.

The problem is that configurator is using MSPv2 messages, but CRSF protocol in inav supports only MSPv1 in MSPOverCRSF implementation.

Lethal-Paw, 2023-01-16:
> Actually this is firmware problem. There are no changes required in configurator.
>
> ELRS is encapsulating MSP messages which are sent to tcp port by configurator into CRSF packets.
>
> The problem is [...]

Aha so my thought wasn't so wrong. That also explains why I can't connect to the Betaflight Configurator even though in was already Implemented in the BF Configurator.
What would you recommend to doabout the issue (since it has been closed and is apparently not being worked on anymore?)

MrD-RC, 2023-01-18:
Regardless of firmware or configurator. An issue already exists for this feature request. That issue should be used going forward, No need to use the duplicate issue.