iNavFlight/inav #8701: Matek F405 WMN and Jeti Receiver - how to setup connection

boomerschnecke, 2023-01-13:
Trying to setup my Jeti with the Matek F405 WMN
Connecting the Jeti with SBUS or JETIEXBUS to the WMN Board to Pin SBUS is not working.

At INAV the Port UART2 and RX= Serial RX

But nothing is working on Jeti TX-RX
The FC itself works, meaning the Servos are working, GPS is working and the Motor could be activated by INAV

Could anyone give me some hints how to connect the Jeti Receiver to the Matk F405 WMN?

****

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

RoadyFPV, 2023-02-05:
Connect your Reciever to UART TX!
Setup your RX port you are using to EX BUS
Choos Serial RX in configurator on the UART u have connected the RX 
Setup Serial Reciever and provider is JETI EXBUS

RoadyFPV, 2023-02-05:
[How to setup your RX and FC for Jeti Ex-Bus.pdf](https://github.com/iNavFlight/inav/files/10611019/How.to.setup.your.RX.and.FC.for.Jeti.Ex-Bus.pdf)

maybe helpful

closed 2023-03-06 by DzikuVx

RobinatiNav, 2023-03-25:
Hi boomerschnecke, I am interested how it went. Did you manage to use JETIEXBUS in iNav? If so, is it reliable? Do you fly it often? Many thanks for your answers.

boomerschnecke, 2023-03-26:
Hi RobintiNav, 
it was working successfully. But not very stable in my eyes.
You have to do an resisitor with 1.2k within Rx und Tx at the FC. Look into Jeti pages. There you find an description.
All telemetry was working well. But the reaction from tranmitter to FC was not very stable. Or sometimes it seems to be some delay within. 
So I switched back to SBUS. This works well up to now

RobinatiNav, 2023-03-26:
Hi, thank you :) check this #7672

RoadyFPV, 2023-03-26:
> Hi RobintiNav, it was working successfully. But not very stable in my eyes. You have to do an resisitor with 1.2k within Rx und Tx at the FC. Look into Jeti pages. There you find an description. All [...]

marv-t optimized the code on betaflight, so no resistor is necessary any more. after that optimation no problems at all.
than he implemented the code to inav... no problem.

After INAV switched to GCC10 the issue still exists..
Same happens to betaflight after switch to gcc10..

RoadyFPV, 2023-03-26:
Pls read also

Line 35

https://github.com/iNavFlight/inav/blob/master/src/main/rx/jetiexbus.c

boomerschnecke, 2023-03-26:
Thanks for the information
if the resistor is not necessary any more, how to do the connection?
Could the SBUS connection than be used for JetiEX Bus connection? Or how to connect this physically correctly?

RoadyFPV, 2023-03-26:
no only solder your wire coming from your Reciever directly to the UART TX PIN on the FC thats all

RobinatiNav, 2023-03-26:
- connect receiver signal cable to Tx port.
- choose EXBUS for serial port in device Explorer
- 
[image]
[image]

boomerschnecke, 2023-03-26:
Thank you for information

boomerschnecke, 2023-04-16:
Hi RobinatiNav, I would like to know, if the system really works if you only connect the Tx from the FC to the Jeti receiver. Jeti Telemetry should work fine, but what about the stick commands the receiver is transferring to the FC? 
An connection to Rx of the FC should also be necessary? Or not?
Best regards

RoadyFPV, 2023-04-16:
> [How to setup your RX and FC for Jeti Ex-Bus.pdf](https://github.com/iNavFlight/inav/files/10611019/How.to.setup.your.RX.and.FC.for.Jeti.Ex-Bus.pdf)
> 
> maybe helpful

boomerschnecke, 2023-04-16:
Thanks for the quick response.
I was just wondering because its a bidirectional communication line but only Tx is used......
So this works for telemetry of Jeti AND with Jeti Transmitter stick commands?

boomerschnecke, 2023-04-16:
I will try it
Thank you

RoadyFPV, 2023-04-16:
Jea it works fine, until inav is using GCC10. 
Only use GCC9 comiled versions for flights

boomerschnecke, 2023-04-16:
Ok, but sorry, how can I do this? Do I have to compile myself? Up to now I´m using the files from the INAV Downloads directly.

RoadyFPV, 2023-04-16:
U can flash a version and check with cli command "version" the gcc version is it not GCC10 its fine.
If u are able to compile it yourself u can use the newest version of inav

boomerschnecke, 2023-04-17:
So checked it: Version is GCC-10.2.1 
So this means I would have to compile the source code by myself with GCC9 ?
Do you know the background why it works with GCC9 and not with GCC10?

boomerschnecke, 2023-04-17:
can you please send a download link to me for the compiler? Is it free?

RoadyFPV, 2023-04-17:
The FC is freezing. we dont  know why but with gcc9 there is no issue.

for compiling:
https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows%2010%20or%2011%20with%20MSYS2.md