iNavFlight/inav #8705: MSP_OSD : DJI O3 char translation table improvement for  INAV: 6.0.0

Sil20, 2023-01-15:
## Current Behavior
using a DJI O3 air unit, the 3D speed KPH unit uses '?'char 
vertical-ish horizon uses '?'char  as well 

## Steps to Reproduce
Enable 3D speed unit and articifial horizon using a DJI O3 air unit

## Expected behavior
Display a proper character 

## Suggested solution(s)
I Tested on a personal build the following solution:
\src\main\io\displayport_msp_bf_compat.c
    case SYM_AH_V_START:
    case (SYM_AH_V_START+1):
    case (SYM_AH_V_START+2):
    case (SYM_AH_V_START+3):
    case (SYM_AH_V_START+4):
    case (SYM_AH_V_START+5):
        return 0x3A;  // 0x7C look like '|'    0x3A is ':' 

    case SYM_3D_KMH:
        return BF_SYM_KPH;

MrD-RC, 2023-01-15:
All elements not supported in the BetaFlight font will use a **?**. This is a known limitation of this hack. If you want full OSD support, you should complain to DJI, asking them to implement full MSP DisplayPort support on their goggles, with custom font uploads. This is the **only** way to get a decent working OSD. Other than a group like WTFDJI hacking the new system.

- IMHO, 3D speed should not be substituted with a non-3D speed character.
- The BetaFlight font does not have vertical AHI characters.

Sil20, 2023-01-15:
I agree with you an know very well the limitation of this and also very well WTFOS.
But the purpose of this displayport_msp_bf_compat feature is to get a decent OSD for DJI O3 , until WTFOS ou DJI implement a full featured OSD  assuming it will happen at some point.

So my points are : 
- Is it better to see "35km/h" instead of "35?" ?
- Is it better to get ':' or '|' than '?' as a vertical ish horizon ?
- Is n't there already some workaround to manage some discrepencies between BF and Inav OSD?

I'm afraid the answer to all toses points is yes !

mmosca, 2023-01-15:
The goal is to be a stop gap measure to at least get something close to useful.  It will never be 100%. 
We may be able to replace the vertical ahi characters with one of the horizontal ones, to get a behavior closer to betaflight, but units can make things confusing. 

If your argument is that you know what field is where, it is not better than the question mark.

mmosca, 2023-01-17:
@Sil20 I created a pull request with changes to the artificial horizon, in case you would like to test it.

closed 2023-01-17 by DzikuVx

Sil20, 2023-01-17:
> The goal is to be a stop gap measure to at least get something close to useful. It will never be 100%. We may be able to replace the vertical ahi characters with one of the horizontal ones, to get a [...]

Thanks for the PR.
Actually my argument was "having a proper speed unit is better than nothing", and "having no speed unit at all is better than having '?' especially if I know where the field is or If I display only 3D speed"

And since we, or at least I don't know how long we will have to wait for a good MSP display port implementation,  I would rather have something better.

Thanks again for your work

PR #8709 (2023-01-17): Use BF SYM_AH_BAR9_4 as replacement for INAV SYM_AH_V_START family in BF Compatibility OSD mode
Symple change that avoids the artificial horizon from changing into a column of `?`.

With this we can fix #8705 .

For the other unit characters we are probably better off leaving them as question marks.