iNavFlight/inav #8716: Home Arrow direction - option to point up = towards home (old behaviour)

wilco1967, 2023-01-18:
Perhaps related to #8586  or #8529 
Airplane.
iNav 6.0.0 compiled from master 03-01-2023

In the current Master 6.0.0, the home arrow on OSD is pointing (correctly) to home, even if the plane is affected by a strong crosswind.

In earlier versions, the arrow would be pointing straight up, if the course over ground was directly towards home. So it was very easy to see if going in the right direction. It didn't matter where the nose was pointing due to crosswind.

In the current master however, the arrow takes into account the heading of the plane (where the nose is pointing), rather then course over ground.
From the perspective of the pilot (or camera), it points to where home actually is,

Although the current behaviour is correct, it is not as easy to use for orientation as it used to be in the old style
It is not always clear, while flying, where the wind comes from., or how strong it is. So difficult to tell what is the straightest line home.

I had a flight last week, where the planeI was above clouds, while there was quite a strong wind. (>30 km/h)
After some time, I noticed the plane wasn't where I assumed it to be. it has flewn out of the antenna beam so I had to redirect the antennas. 
It had drifted quite some distance downwind.

After hitting RTH, after it made the turn, I noticed, the home arrow wasn't pointing up as it used to do, but was pointing quite a lot to the left. 
At that moment, I really thought something was wrong (didn't realize at the time it was actually pointing to where home was, instead of straight up for home). 
It seemed like it was making a large turn while coming home. (couldn't see the ground at the time).

Later, I called up the telemetry log, and noticed, that it had in fact flew a perfect straight line back home. It was just being affected by the strong crosswind. So everything was correct, and it did what it needed to do. 

The only 'problem' is, this isn't as easy to use for orientation, as the old style where 'up' was flying towards home (even if the nose is pointing in a slighly different direction due to crosswind).

Perhaps we can have an option to show either the home arrow relative to the current heading (where the nose is pointing), or the old style, where it is relative to course over ground. (or even both options).

Thanks a lot !

breadoven, 2023-01-19:
Previously, if there was no compass, heading was in fact just ground course. Now heading is calculated from wind estimations so can be different to ground course. I'm guessing the home arrow direction is using heading as its reference  which is no longer correct when there's wind causing a difference between heading and ground course. Probably just needs changing to use ground course on a fixed wing to fix this issue.

breadoven, 2023-01-20:
Looking at the code it is using heading to calculate the arrow direction. Needs double checking but this probably should be changed to ground course for a fixed wing so a fix likely required.

wilco1967, 2023-01-20:
Great, Thanks !

Perhaps, an option (CLI parameter) to let the user select to either use 'ground course' or 'heading' for the home arrow would be ideal ?

I could understand some people preferring the new way instead.  After all, it is pointing to home correctly (just not always straight up for a straight flight home).
Personally, I liked the old method better.

breadoven, 2023-01-21:
I think the home arrow should always point directly up screen when you are heading directly toward home. There is also OSD home heading error which shows the error between the current course and direction to home. This should be showing 0 when heading directly toward home which currently it isn't with a fixed wing when wind is present, it's off slightly the same as the home heading arrow. I've already tested a fix that seems to fix this, PR to follow.

closed 2023-01-21 by DzikuVx