iNavFlight/inav #8730: Improved angle based flight modes with Feedforward and coordination

Jetrell, 2023-01-24:
## Current Behavior
Whether using Angle or Cruise/Course Hold modes. They aren't very responsive to stick commands, when comparing to Acro mode. Both for Multirotor and Fixedwing platforms.
Attitude stabilized flight modes are very useful, but they aren't always practical for more advanced pilots.
Now that INAV's AHRS has drift correction.. I think this feature would make a large improvement to Angle based flight modes.

Presently, two guys have been working on a fix for this in Betaflight. With very promising results.

https://github.com/betaflight/betaflight/pull/12081

Chris Rosser's video.
https://www.youtube.com/watch?v=ILeLo1lWjBk

b14ckyy, 2023-01-31:
I think I understand your basic idea but maybe it should not be as radical as betaflight does it, especially on fixed wings. 

better axis coordination might be helpful on quads indeed, that's a good point. 

The CD (I assume they actually added a control derivative and not FF) should only be applied to ANGLE and HORIZON though and not to nav modes. 
Turns in nav modes create a heading error sent to the attitude controller, that creates a gyro setpoint for the plane and initiates the banking. if you would implement a CD here, navigation turns would be sharp and snappy. This would completely interfere with control smoothness and make navigation and WP flights look harsh and agressive. 

Alttiude changes in nav modes are controlled by vertical speed changes and the vspeed error is created by the altitude error afaik. This is necessary for achieving constant climb rates and straight climbs. CD for the pitch angle might cause trouble with the vspeed PID controller as well and navigating purely by pitch angle is also unusable as it will be impossible to achieve a constant climb rate. 

So if optimizations are made here it should exclusively affect ANGLE and HORIZON modes and not navigation modes. For Nav modes other optimizations are needed in the future.

b14ckyy, 2023-01-31:
> While I dislike the feel of Angle mode, which requires almost full stick deflection to alter the attitude of the craft.. 

This can be overcome by either raising the bax bank angle and max pitch angle to get more movement overall but also by raising the angle strength slightly to get higher rate setpoints for the attitude control. But the latter can cause oscillations if PIDFF is not tuned properly. But maybe a expo curve for Angle mode might be helpful indeed to get more reaktion at slight stick movements without altering the max bank angle. 

> The reason I know, Is because I like Cruise mode, but loath how slow Cruise mode reacts to stick input, if a fast response is required.

this could be resolved actually pretty easily. Quads have the option of ATTITUDE or CRUISE mode to control the position in POSHOLD. 
Attitude temporarily switches to ANGLE when stick is touched and back into POSHOLD when stick is centered, while with CRUISE control type, the position based on GPS is altered by the stick input and the multirotor just corrects its holding position. 

Something similar could be done on wings I guess so when stick is touched, althold and heading are disabled and the pilot has direct atttitude control as in angle mode and as soon as the stick is centered, course and altitude hold is reengaged. Maybe even getting a HORIZON like experiece. 

So the control mode for Cruise mode (or course hold as well) could be CRUISE, ANGLE or HORIZON when stick input is given. Could be easily tested with logic conditions to see how thze plane behaves.

Jetrell, 2023-01-31:
@b14ckyy I'm aware of the workarounds.. But they're just that... What a workaround **partially** enhances in one area, it messes up something worse in another.  Its always a tradeoff, and is never a substitute for a purpose designed feature.

I opened this request, simply because both of Chris's fixes are the way Angle mode should have been designed to work in the beginning, in the days of Baseflight.

b14ckyy, 2023-01-31:
I am not explaining workarounds. these are suggestions for feature changes how your ideas could be achieved. 😉

Jetrell, 2023-01-31:
>This can be overcome by either raising the bank bank angle and max pitch angle to get more movement overall but also by raising the angle strength slightly to get higher rate setpoints for the attitude control. But the latter can cause oscillations if PIDFF is not tuned properly. But maybe a expo curve for Angle mode might be helpful indeed to get more reaktion at slight stick movements without altering the max bank angle.

I was more referring to your statement here.

closed 2023-12-28 by Jetrell