iNavFlight/inav #8733: SpeedyBee F405 V3 uSD card vs OSD interference

Pedrolomel, 2023-01-25:
Hello
       I found issue with OSD interference(or even OSD disappear) when uSDHC card(4GB) is inserted for blackbox logging. Tested for two FCs copies and different uSDHC cards. 

FC: SpeedyBee F405 V3
FW: iNAV 5.1

Pedrolomel, 2023-01-27:
Hellow

Some screenshots documenting current issue:

Without uSD card:

[image]
 
uSD card inserted:

[image]

Phil-MC, 2023-01-28:
How was it formatted? 
Did you follow speedybee or Inav guidlines for formatting?

Pedrolomel, 2023-01-28:
Yes, but It doesn't matter. I have formatting using SDcard tool from SD org. page.

Phil-MC, 2023-01-28:
OK, it makes the information more complete. Windows 10 doesn't offer it any more. Sent from Outlook for Android

girgine, 2023-01-28:
Hi Pedrolomel, I have the same FCC but after flashing 5.1 , I cant access to the FCC (USB time-out error), is there a specific instruction to flash this FCC with INAV? did you use the standard firmware that came with INAV 5.1?
thanks

Phil-MC, 2023-01-28:
You must use the firmware name provided by speedybee on their website. I did, and it worked . Did you choose v3?

girgine, 2023-01-28:
they just point to inav 5 to their web site, I didnt see a firmware file , was there? (checking it again..) thanks
I did choose v3 from inav 5.1 flasher, it does the flash without an issue but then cant connect to FCC.. (if chose only F405, it can connect but its not my fcc ..)

Pedrolomel, 2023-01-28:
> You must use the firmware name provided by speedybee on their website. I did, and it worked . Did you choose v3?

Yes. I have chose firmware for target SPEEDYBEEF405V3.

Pedrolomel, 2023-01-28:
> they just point to inav 5 to their web site, I didnt see a firmware file , was there? (checking it again..) thanks I did choose v3 from inav 5.1 flasher, it does the flash without an issue but then cant connect to FCC.. (if chose only F405, it can connect but its not my fcc ..)

Remove SD card (If inserted). With some SD cards,  SpeedyBee F405V3 can't connect to configurator.

girgine, 2023-01-29:
Wow! removed the SD-Card and it worked - cant believe this :(

thanks very much! it seems I need to look for another SD-card then :)

Phil-MC, 2023-01-29:
The file isn't there but the name of the firmware is.SPEEDYBEEF405V3There are issues with the SD CARD. The system uses the card in a way it was not designed for and thus has speed problems. T'm experimenting at the moment.Sent from Outlook for Android

Pedrolomel, 2023-01-30:
> Wow! removed the SD-Card and it worked - cant believe this :(
> 
> thanks very much! it seems I need to look for another SD-card then :)

I'm afraid it doesn't solve problem. I have switched off OSD on iNAV config. tab. After that, all uSD cards has recognized properly by FC and connection problems disappear. I gues, the problem is related  to the conflict between OSD chip and uSD cards. In short, OSD and uSD can't works together.

Phil-MC, 2023-01-31:
That finding is very interesting. Is the OSD interaction with the SD card affected by loading an OSD FONT in the Configurator?Sent from Outlook for Android

Phil-MC, 2023-01-31:
I have tried turning off the OSD myself and the BlackBox log is recorded on an ordinary SD card as you say. So I can cofirm this anomaly.

Pedrolomel, 2023-01-31:
> I have tried turning off the OSD myself and the BlackBox log is recorded on an ordinary SD card as you say. So I can cofirm this anomaly.

Phil-MC, 2023-02-01:
I notice in the Flight Modes section of the speedybee app there are two related ones:
Enable/Disable BlackBox Recording
Enable/Disable OSD
So it is possible to set a Transmitter switch to select one combination or the other.

Pedrolomel, 2023-02-02:
With Betaflight firmware, OSD and blackbox logging simultaneously works fine. In my opinion the fault is on iNAV side.

Phil-MC, 2023-02-02:
INAV requires deep access to the FC in order to provide functions that are not available in BetaFlight. The OSD and BlackBox functions occupy the same internal bus in the design of the FC. This is what causes the issue.Sent from Outlook for AndroidFrom: Pedrolomel ***@***.***>Sent: Thursday, 2 February 2023, 08:11To: iNavFlight/inav ***@***.***>Cc: Philip ***@***.***>; Comment ***@***.***>Subject: Re: [iNavFlight/inav] SpeedyBee F405 V3 uSD card vs OSD interference (Issue #8733)
With Betaflight firmware, OSD and blackbox logging simultaneously works fine. In my opinion the fault is on iNAV side.

Pedrolomel, 2023-02-02:
There is no such issues on MATEK targets with iNAV firmware. if on MATEK targets iNAV works, so It could work on SpeedyBee target....I hope.

shirase, 2023-03-16:
I tried decreasing SDCARD_BUS_SPEED BUS_SPEED_SLOW and max7456Init BUS_SPEED_SLOW. It helped me.

shirase, 2023-03-17:
> I tried decreasing SDCARD_BUS_SPEED BUS_SPEED_SLOW and max7456Init BUS_SPEED_SLOW. It helped me.

I did some tests. And it works absolutely unstable. OSD disappears. After 30 seconds the logs stops. But you can connect to the FC with SD card and OSD.

shirase, 2023-03-17:
If disable DEVFLAGS_USE_MANUAL_DEVICE_SELECT for DEVHW_SDCARD then everything works. 

https://github.com/iNavFlight/inav/commit/8913f6c4366b6eb4853cbefb4c9e9d6ef55c0edc

Phil-MC, 2023-03-17:
Thanks for the update.

shirase, 2023-03-17:
I can't figure out what DEVFLAGS_USE_MANUAL_DEVICE_SELECT is used for. Who knows? What does it affect?

Phil-MC, 2023-03-18:
Thanks for the update.
I don't recognise the CLI entry, where did you get it from?

shirase, 2023-03-18:
This is from the inav code. You can try the compiled version.
https://disk.yandex.ru/d/8qYjAU4mwVU71g

Phil-MC, 2023-03-19:
Thanks, so that wasn't a CLI entry. Was it from target.c ?

Phil-MC, 2023-03-19:
I have verified that the Configurator will not connect unless the Blackbox SD CARD is removed.

Pedrolomel, 2023-03-19:
https://github.com/iNavFlight/inav/issues/8733#issuecomment-1408457424

closed 2023-05-23 by DzikuVx

Phil-MC, 2023-06-11:
I have found that with WalkSnail Avatar set there is no problem with having Blackbox and OSD in use at the same time. I don't know if this is as a result of the fix or because of the MSP Display Port use.

PR #9028 (2023-05-23): Fix: SpeedyBee F405 V3 uSD card vs OSD interference
Fix: #8733 #8766  #8527 #7707 #8884

If SD card and OSD share same SPI bus, like SpeedyBee F405 V3. We can't use DEVFLAGS_USE_MANUAL_DEVICE_SELECT, because some time both devises stay selected. We must use same SPI MODE. And we muse use same SPI bus speed.