iNavFlight/inav #8740: telemetry or gps inexplicable issue

brousselle, 2023-01-27:
## Current Behavior
i tried moving several time from inav 5.0 to inav 6.0 in order to test and try O3 osd, but i always face the same issue:
inav 6.0 waits for gps fix even with >6 sats, and telemetry does not work anymore on my crossfire setup.

i have a matek F405 wse with crossfire nano diversity, bn200 gps, DJI O3 and google 2.

1) inav5 works as expected, with all navigation, rth, telemetry (sats and coordinates on the tango2). see [INAV_cli_20221215_latestInav5.txt](https://github.com/iNavFlight/inav/files/10516333/INAV_cli_20221215_latestInav5.txt) for the quite standard conf i fly with.
2) flashing inav 6 (see [INAV_6.0.0_cli_20230126_test2.txt](https://github.com/iNavFlight/inav/files/10516354/INAV_6.0.0_cli_20230126_test2.txt)): plane seems ok, but i cannot arm, and no telemetry.
- CLI: status looks normal (see screenshot)
- inav configurator show 7 satelites (min is default 6) but still say navigation is not safe. (see screenshot)
- osd also shows 7 sats, but warning message "no gps fix" (see screenshot)
- tango2 does not received any of the sensors that were previously working (doing a new detection does not help) so no number of sats neither coordinates.
3) flashing back inav5 with exact same setup ([INAV_cli_20221215_latestInav5.txt](https://github.com/iNavFlight/inav/files/10516333/INAV_cli_20221215_latestInav5.txt)) makes everything works again.

I tried this full process at least 3 times, with hex build from master (yesterday for the latest) but a few weeks ago for the others.

- no changes in hardware or wiring.
- no changes on tango2 side.
- for some time i have suspected a hardware defect on my FC, but then why reverting to inav5 works ?

This issue looks really weird to me, and i have no clue about how i can debug this further.
Any help appreciated.

[image]
[image]
[image]
[INAV_cli_20221215_latestInav5.txt](https://github.com/iNavFlight/inav/files/10516333/INAV_cli_20221215_latestInav5.txt)
[INAV_6.0.0_cli_20230126_test2.txt](https://github.com/iNavFlight/inav/files/10516354/INAV_6.0.0_cli_20230126_test2.txt)

OptimusTi, 2023-02-03:
Please show us the ports tab

brousselle, 2023-02-05:
you can check the the full conf in my diff output on my original post. 
But here is the screenshot for inav5 (gps and telemetry works there):
[image]

And here is the screenshot for inav6 (gps weird behavior and no telemetry):
[image]

As you can see they are identical except the OSD port.

MrD-RC, 2023-02-05:
You are using the wrong Configurator. You should be using 6.0 RC1. Do not use FP1/FP2 anymore.

brousselle, 2023-02-06:
Indeed, using configurator 6.0 RC1: i have telemetry that now works.
Give me 2 more days, so i can try with the GPS. I will report back here and close the issue if all good !
Thanks for the tip !

brousselle, 2023-02-09:
I confirm, this is all good now: Thanks for your help.

closed 2023-02-09 by brousselle