iNavFlight/inav #8749: VTX power reverts to 1 no matter how it is set version 6.0.0-RC1

miggy1952, 2023-01-30:
In this example I will specifically mention this is on a MATEKF765 but it is the same on others.

Using version 6.0.0-RC1

I tested "Always" and "Until first arm" setting the power level to 4 but upon arming it reverts to power level 1

MrD-RC, 2023-01-30:
More details would be useful.:
- Is this IRC Tramp or SmartAudio?
- Which VTx?
- Was this setup working on 5.1?

I flew on Sunday with SmartAudio on a Rush Tank Solo. I was using what would become RC1 about an hour later, and it worked perfectly. I had power changing based on 3D distance. As well as having low power on disarm.

miggy1952, 2023-01-30:
It is on a Matek VTX-MINI VTX and yes it was working fine on 5.1
Thanks for your help, love your YT channel!

Jetrell, 2023-02-01:
@miggy1952 I tested INAV 6.0RC1 with a Matek VTX mini. And it works fine for me. The levels increase from 1, when armed. Then move through the range upto 5, and return to 1 when disarmed.
If your using logic conditions to dynamically switch between power levels. Be mindful, that there has been reordering of the logic flight operands.

miggy1952, 2023-02-06:
Thanks for your help. Been experimenting and my problem is almost certainly NOT within INAV. Using either INAV or Betaflight configurators, I set the power level and set low power until first disarm and when I arm the power remains at 25mW. This is in both a tramp and or a TBS smart audio VTX. When I use smart audio on the goggles to change the power it works fine, the power goes up as expected. But if I power cycle and try again the problem returns and the power remains at 25mw.

miggy1952, 2023-02-07:
Problem solved, I was using ELRS and once I disabled the vtx administrator everything was good again.

closed 2023-02-07 by miggy1952