iNavFlight/inav #8750: Add MSP VTX functionality

MartinHugh, 2023-01-30:
## Current Behavior
Currently VTX control is available via Smart Audio.   This requires a dedicated UART
Betaflight and Quicksilver now use MSPVtx (for OpenVTX and HDZero) which uses the MSP UART to control the VTX.

## Desired Behavior
Implement MSP VTX, so that MSP protocols encompass VTX control, thereby saving a UART

## Who does this impact? Who is this for?
Any iNav user who may want Smart Audio commands without needed to use a UART

sebas1984x, 2023-05-06:
We need this!

robhaswell, 2023-05-30:
There is also the 500mW double-stack board, which is great for wings, that does not support Smart Audio.

DerWolle82, 2023-06-08:
We realy need this!

MartinHugh, 2023-06-11:
Another strong requirement for this is that ELRS Backpack (which allows users to change channels on both goggles and aircraft, sumultaneously, from a LUA command), is not usable with iNav.  

This is because the part which commands the vtx to switch channels requires MSP-VTX.

MrMap, 2023-07-27:
I must admit that it took me some time of trying and trying to control the HD Zero Freestyle VTX via the UART, only before I had to realize that it could not work without SA (simplex and one extra UART). 
I thought well I remember that years ago TBS has VTX control over UART for analog VTX, surely the same concept (now MSP VTX) is available in iNav 6.1?! Nope. 
Well - I hope this is on the devs prio list - it would be so nice.

sebas1984x, 2023-07-28:
> I must admit that it took me some time of trying and trying to control the HD Zero Freestyle VTX via the UART, only before I had to realize that it could not work without SA (simplex and one extra [...]

https://github.com/hd-zero/hdzero-vtx/pull/141

DzikuVx, 2023-08-01:
Added in #9166

closed 2023-08-01 by DzikuVx