iNavFlight/inav #8754: Apparently random current spikes on 7 inch and heading(yaw) > 360°

moserello, 2023-01-31:
## Current Behavior
Hello all, and thanks in advance for the time used helping me to sort this problem out.

I've a 7inch quad, with the stack Kakute F7 mini v3 + Tekko32 F3 4in1 mini 45A, that generally flies very well. 
The current sensor is well calibrated, FC and motors softmounted. 
FW = INAV 5.1 
log FM = ACRO

The problem is that, apparently in random moments, the current consumption increases to incredible high readings: it jumps from 6-8A of easy cruising to even 100A. The battery voltage drops and it is possible to perceive the sag of the battery on the drone responses. 
The Kakute F7 mini v3 has just 16MB of free memory for blackboxing ( Poop ), so it was not so easy to see those spikes on logs.
I attach a log file where the current absorption goes to 80A, but I don't see what causes this.

[LOG FILE](https://www.mediafire.com/file/0f8pihzxbyogpzr/blackbox_log_2023-01-27_184731.TXT/file)

Moreover, I saw another weird thing: the heading evaluated by INAV (heading[yaw]) that should range from 0° to 360°, goes to over 400° in some moments ! why? what does it mean? can the 2 problems can be correlated?

Here's my diff:

```
diff

# version
# INAV/KAKUTEF7MINIV3 5.1.0 Aug 19 2022 / 11:42:00 (76f22b25)
# GCC-10.2.1 20201103 (release)

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mixer

# servo

# safehome

# feature
feature GPS
feature CURRENT_METER
feature PWM_OUTPUT_ENABLE

# beeper

# blackbox
blackbox NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS

# map
map TAER

# serial
serial 0 0 115200 115200 0 115200
serial 1 2048 115200 115200 0 115200
serial 3 2 115200 57600 0 115200
serial 5 64 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1800 2100
aux 1 2 5 1300 1675
aux 2 11 5 1800 2100
aux 3 10 3 1800 2100
aux 4 52 4 1800 2100
aux 5 13 2 1600 2100
aux 6 27 1 1800 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 26 14 H
osd_layout 0 1 12 0 H
osd_layout 0 3 8 6 V
osd_layout 0 6 23 1 V
osd_layout 0 7 13 12 H
osd_layout 0 9 9 13 V
osd_layout 0 11 1 2 V
osd_layout 0 12 15 14 V
osd_layout 0 13 17 0 H
osd_layout 0 14 25 2 V
osd_layout 0 15 8 1 V
osd_layout 0 20 12 15 V
osd_layout 0 21 1 15 V
osd_layout 0 22 23 14 V
osd_layout 0 23 24 14 V
osd_layout 0 24 12 2 V
osd_layout 0 28 23 0 H
osd_layout 0 30 1 10 V
osd_layout 0 32 1 1 V
osd_layout 0 85 16 1 V
osd_layout 0 86 23 15 V
osd_layout 0 97 1 14 V
osd_layout 0 110 1 13 V

# logic

# global vars

# programmable pid controllers

# master
set looptime = 500
set gyro_main_lpf_hz = 110
set gyro_main_lpf_type = PT1
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 120
set setpoint_kalman_q = 200
set gyro_zero_x = -75
set gyro_zero_y = -6
set gyro_zero_z = 3
set ins_gravity_cmss =  985.615
set acc_hardware = MPU6000
set acczero_x = 54
set acczero_y = -12
set acczero_z = -156
set accgain_x = 4085
set accgain_y = 4104
set accgain_z = 4001
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set magzero_x = -98
set magzero_y = -483
set magzero_z = 534
set maggain_x = 382
set maggain_y = 363
set maggain_z = 480
set align_mag_roll = 150
set align_mag_pitch = 1800
set align_mag_yaw = 800
set baro_hardware = BMP280
set serialrx_provider = CRSF
set blackbox_rate_denom = 2
set motor_pwm_rate = 8000
set motor_pwm_protocol = DSHOT600
set failsafe_procedure = RTH
set failsafe_min_distance = 2000
set failsafe_min_distance_procedure = LAND
set current_meter_scale = 82
set motor_direction_inverted = ON
set model_preview_type = 3
set small_angle = 180
set applied_defaults = 2
set rpm_gyro_filter_enabled = ON
set rpm_gyro_min_hz = 70
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set gps_min_sats = 5
set airmode_type = THROTTLE_THRESHOLD
set inav_w_z_baro_p =  0.350
set nav_extra_arming_safety = OFF
set nav_rth_climb_ignore_emerg = ON
set nav_rth_alt_mode = AT_LEAST_LINEAR_DESCENT
set nav_rth_altitude = 3000
set nav_fw_launch_max_angle = 180
set osd_video_system = PAL
set osd_crsf_lq_format = TYPE2
set osd_main_voltage_decimals = 2
set i2c_speed = 800KHZ
set name = Rekon7 LR
set tz_automatic_dst = EU
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 4
set vtx_low_power_disarm = UNTIL_FIRST_ARM

# profile
profile 1

set mc_p_pitch = 60
set mc_i_pitch = 45
set mc_d_pitch = 29
set mc_p_roll = 55
set mc_i_roll = 38
set mc_d_roll = 27
set mc_p_yaw = 60
set mc_i_yaw = 65
set dterm_lpf_type = PT3
set d_boost_max =  1.500
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set smith_predictor_delay =  1.500
set tpa_rate = 20
set tpa_breakpoint = 1200
set rc_yaw_expo = 70
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60

# battery_profile
battery_profile 1

set bat_cells = 6
set vbat_min_cell_voltage = 300
set vbat_warning_cell_voltage = 330
set throttle_idle =  7.000
set nav_mc_hover_thr = 1350

# save configuration
save

# end the command batch
batch end

# 
```

I have no idea about what can cause these issues, any help is greatly appreciated
Thanks in advance!  

## Steps to Reproduce
1. Plug the lipo
2. Fly 

## Expected behavior
Fly without random current spikes

# INAV/KAKUTEF7MINIV3 5.1.0 Aug 19 2022 / 11:42:00 (76f22b25)
# GCC-10.2.1 20201103 (release)

b14ckyy, 2024-03-20:
there is a tuning issue. you have a massive amount of motor noise. 
Normal flight at low amps: 
[image]

then a spike comes in with barely any movement or throttle increase: 
[image]

Seems to be your Pitch D-Term: 
[image]

closed 2024-03-20 by b14ckyy

moserello, 2024-03-22:
@b14ckyy thanks a lot for taking the time checking this log. Mounting rigidity bars on the frame (not available in a previous moment), and lowering Dterms solved the problem