iNavFlight/inav #8758: LEDs not working on Matek F405-WSE

Andy0679, 2023-02-01:
When trying to set LEDs on LED tab in Configurator 6 RC 1 the squares don't show a number now stay selected.

cmanley, 2023-02-06:
I've got the same firmware and board and the LEDs work for me.
My settings for wing tip nav lights:
```
feature LED_STRIP
led 0 0,0::CIWE:2
led 1 1,0::CIWE:6
```

Helipro, 2023-02-23:
> When trying to set LEDs on LED tab in Configurator 6 RC 1 the squares don't show a number nor stay selected.
Or you using orientation tags?  I couldnt get leds to save if orientations were selected in 6.1 or .6.2.
https://github.com/iNavFlight/inav/issues/8823

dzhibas, 2023-10-06:
i do have configurator 6.1.0 with matek f405-wse and led strip shows me "0 remaining" but through CLI with commands like
```
led 0 0,0::CIWE:2
led 1 1,0::CIWE:6
etc...
```

stuff works. why is it "0 remaining" in UI and not able to select leds

dzhibas, 2023-10-07:
Oh, i'v upgraded to latest target and LED Strip started to work:
[image]
[image]

i have no issues anymore

closed 2024-03-22 by b14ckyy