iNavFlight/inav #8762: RSSI OSD display in iNAV 6.0.0 (RC1)

NorthIceMan, 2023-02-03:
Could not get RSSI to show anything but "1" (number 1)  So in CLI  I set "rssi_source = CHANNEL", and "rssi_channel = 11".  Then in Receiver page I set "RSSI Channel" to 11.   In Taranis Qx7 I Config an Input (for rssi) with source on RSSI , then an Output to send rssi to channel 11.  RSSI then shows on OSD as 99 (Oh well, missing only 1% is not bad!).

FYI-I am connecting to a Matek H743 WING FC via FrSky R9SX receiver and R9M-2019 module on a Qx7 - configured as FPORT.
BTW-in Receiver page both "Serial Port Inverted (comparing to protocol default)" AND "Serial receiver half-duplex" had to be set to OFF
I wondered if the 100% (true rssi value) could be reached by somehow multiplying that "1" by 100 someplace? ... and why FPORT doesn`t get this to FC ?
If there is a more elegant way to do this by manipulating the rssi settings in CLI please let us know.

OptimusTi, 2023-02-03:
For some reason that escapes my sleepy mind, RSSI goes 100 units but it starts from 0 to 99. Other protocols like crsf, don't have this issue because rssi is scaled differently.

NorthIceMan, 2023-02-10:
>Correction....in the above...
 "then an Output to send rssi to channel 11."   SHOULD READ "then an entry in MIXES for Input rssi on Cnannel 11.

b14ckyy, 2024-03-21:
if you use FPort, then in CLI just use `set rssi_source = PROTOCOL`
you need to correct the scaling as fport scales from 0-49 so you need to `set rssi_max = 50` No channel assignment needed and you should be golden.

closed 2024-03-22 by b14ckyy

NorthIceMan, 2024-03-23:
Thanks for your reply.
Is it same if I use SBUS?

Ron

sensei-hacker, 2024-03-23:
> Thanks for your reply. Is it same if I use SBUS? Ron

F.Port really just packages Sbus and Smartport together.
So yeah, f.port *is* Sbus.

b14ckyy, 2024-03-23:
Although f.port contains a pure sbus stream, the RSSI feedback comes over the (already bidirectional) SmartPort stream.

When you use sbus + SmartPort Telemetry, then SmartPort is handled as a one way street and RSSI does not come back in protocol form. Then you need a RC channel to provide RSSI info to the FC.

When using f.port then the SmartPort datastream is bidirectional and you get RSSI as a data package to the FC.

Maybe this is a good point for an enhancement?

NorthIceMan, 2024-03-23:
Thanks for the clarification and yes it probably wouldn't take much work to
get to an enhancement.

I have been building and flying model aircraft for 71 years and there is
probably more that I forgot than what I remember !

I'm in NB Canada - where are you located?
Do you fly drones or airplanes?

Ron