iNavFlight/inav #8765: OSD Freeze with DJI WTFOS when bit rate gets low

BogieGirl, 2023-02-03:
## Current Behavior
 When bit rate does down very low - the OSD freeze, toggle to osd #2 will not affect - the OSD is stuck!!. the issue does't happen to my friends flying the same setup with ArduPilot 

Resetting Air unit or disarming the FC unfreeze it!

## Steps to Reproduce

1. bitrate goes down because of someone is on the safe frequency or long distance
2. OSD freezes 
3. the WTFOS ver is 0.9.1 ( latest) and msp-osd compress_osd = true

## Expected behavior
the OSD shall never freeze

## Suggested solution(s)
some way to reset the issue ( toggling to 2nd osd will not respond)

# version
# INAV/MATEKF405SE 6.0.0 Oct 20 2022 / 15:20:51 (d1f6f19a)
# GCC-10.2.1 20201103 (release)

# start the command batch
batch start

# reset configuration to default settings
defaults noreboot

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# servo mixer
smix reset

smix 0 1 0 100 0 -1
smix 1 2 0 100 0 -1
smix 2 3 1 50 0 -1
smix 3 3 2 50 0 -1
smix 4 4 1 -50 0 -1
smix 5 4 2 50 0 -1
smix 6 5 15 100 0 -1

# servo
servo 1 1000 2000 1439 -100
servo 2 1000 2000 1439 -100
servo 3 1000 2000 1526 -100
servo 4 1000 2000 1473 -100
servo 5 600 2100 1480 -100

# safehome

# feature
feature -TX_PROF_SEL
feature MOTOR_STOP
feature GPS
feature PWM_OUTPUT_ENABLE

# beeper
beeper -RUNTIME_CALIBRATION
beeper -MULTI_BEEPS
beeper -SYSTEM_INIT
beeper -ON_USB

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox -GYRO_RAW
blackbox -PEAKS_R
blackbox -PEAKS_P
blackbox -PEAKS_Y

# map

# serial
serial 0 1 115200 115200 0 115200
serial 3 2 115200 115200 0 115200
serial 4 33554432 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 3 1300 2100
aux 1 1 2 1550 1700
aux 2 12 2 900 1100
aux 3 53 2 1200 1400
aux 4 10 2 1900 2100
aux 5 21 5 1800 2100
aux 6 54 5 1400 1600
aux 7 42 1 1925 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 28 19 V
osd_layout 0 1 12 19 V
osd_layout 0 2 0 0 V
osd_layout 0 3 8 6 V
osd_layout 0 7 27 20 V
osd_layout 0 8 0 14 H
osd_layout 0 9 1 12 V
osd_layout 0 11 1 11 V
osd_layout 0 12 18 20 V
osd_layout 0 13 19 11 V
osd_layout 0 14 33 19 V
osd_layout 0 15 35 11 V
osd_layout 0 20 0 20 V
osd_layout 0 21 0 19 V
osd_layout 0 22 30 2 V
osd_layout 0 23 1 17 V
osd_layout 0 24 37 0 V
osd_layout 0 25 40 9 V
osd_layout 0 26 19 12 V
osd_layout 0 27 1 3 V
osd_layout 0 28 38 20 V
osd_layout 0 30 16 17 V
osd_layout 0 31 33 20 V
osd_layout 0 32 12 20 V
osd_layout 0 33 8 3 H
osd_layout 0 34 26 0 V
osd_layout 0 38 18 19 V
osd_layout 0 41 36 12 V
osd_layout 0 46 1 4 V
osd_layout 0 47 1 5 V
osd_layout 0 110 21 16 H
osd_layout 1 7 23 16 H
osd_layout 1 16 5 4 V
osd_layout 1 17 5 3 V
osd_layout 1 18 5 5 V
osd_layout 1 30 12 13 H
osd_layout 1 63 5 11 V
osd_layout 1 67 5 8 V
osd_layout 1 68 5 9 V
osd_layout 1 69 5 10 V
osd_layout 1 119 5 1 V

# logic

# global vars

# programmable pid controllers

# master
set gyro_main_lpf_hz = 25
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 30
set gyro_zero_x = -12
set gyro_zero_y = -16
set gyro_zero_z = 12
set ins_gravity_cmss =  989.909
set acc_hardware = MPU6000
set acczero_x = 56
set acczero_y = -19
set acczero_z = -252
set accgain_x = 4088
set accgain_y = 4116
set accgain_z = 4066
set align_mag = CW270FLIP
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = MS4525
set rssi_channel = 16
set motor_pwm_protocol = STANDARD
set failsafe_procedure = RTH
set align_board_roll = 15
set current_meter_offset = 250
set has_flaps = ON
set model_preview_type = 28
set small_angle = 180
set imu_acc_ignore_slope = 10
set imu_inertia_comp_method = ADAPTIVE
set applied_defaults = 3
set gps_sbas_mode = AUTO
set gps_ublox_use_galileo = ON
set airmode_type = STICK_CENTER_ONCE
set nav_wp_radius = 5000
set nav_rth_allow_landing = NEVER
set nav_rth_altitude = 5000
set nav_rth_home_altitude = 8000
set nav_fw_loiter_radius = 4000
set nav_fw_control_smoothness = 2
set nav_fw_launch_motor_delay = 100
set nav_fw_launch_max_altitude = 5000
set nav_fw_launch_climb_angle = 25
set nav_fw_allow_manual_thr_increase = ON
set osd_video_system = DJIWTF
set osd_time_alarm = 25
set osd_alt_alarm = 1000
set osd_dist_alarm = 3000
set osd_current_alarm = 15
set osd_left_sidebar_scroll = SPEED
set osd_right_sidebar_scroll = ALTITUDE
set osd_coordinate_digits = 8
set osd_switch_indicators_align_left = OFF
set dji_use_name_for_messages = OFF

# profile
profile 1

set fw_p_pitch = 15
set fw_i_pitch = 10
set fw_d_pitch = 5
set fw_ff_pitch = 180
set fw_p_roll = 15
set fw_i_roll = 3
set fw_d_roll = 7
set fw_ff_roll = 140
set fw_p_yaw = 50
set fw_i_yaw = 0
set fw_d_yaw = 20
set fw_ff_yaw = 255
set max_angle_inclination_rll = 450
set max_angle_inclination_pit = 600
set dterm_lpf_hz = 10
set fw_turn_assist_pitch_gain =  0.400
set nav_fw_pos_z_p = 25
set nav_fw_pos_z_d = 7
set nav_fw_pos_xy_p = 55
set d_boost_min =  1.000
set d_boost_max =  1.000
set fw_level_pitch_trim = -1.000
set rc_expo = 30
set rc_yaw_expo = 30
set roll_rate = 9
set pitch_rate = 7
set yaw_rate = 2

# profile
profile 2

# profile
profile 3

set fw_p_roll = 140
set fw_ff_roll = 140

# battery_profile
battery_profile 1

set vbat_warning_cell_voltage = 340
set throttle_idle =  5.000
set nav_fw_max_thr = 1550

# battery_profile
battery_profile 2

# battery_profile
battery_profile 3

# restore original profile selection
profile 1
battery_profile 1

# save configuration
save

# 
---

- FC Board name and vendor:
MATEKF405SE 
- INAV version string:
- INAV/MATEKF405SE 6.0.0 Oct 20 2022 / 15:20:51 (d1f6f19a)

Jetrell, 2023-02-03:
You could try loading INAV 6.0RC1... It has `osd_msp_displayport_fullframe_interval` OSD refresh timer.

Sometimes, the lockup issue you are referring to, is related to the DJI hardware. I have one DJI Air unit, that experiences some strange OSD corruption.. While the other ones work fine, with that same WTFOS and INAV firmware.

BogieGirl, 2023-02-04:
> You could try loading INAV 6.0RC1... It has `osd_msp_displayport_fullframe_interval` OSD refresh timer.
>
> Sometimes, the lockup issue you are referring to, is related to the DJI hardware. I have one [...]

Yes, but my field friends have the same setup just ardupilot and they never experience that issue !

MrD-RC, 2023-02-04:
If you are on 6.0 FP, you should update to RC 1 for both the firmware and Configurator anyway. There are a lot of improvements, including with the MSP DisplayPort OSD.

Packet loss is not something INAV has anything to do with. That should be handled by the video hardware. Both BetaFlight and ArduPilot send all OSD data with every frame. So it is not that the link is working any differently. But the next frame to make it through contains all the OSD elements. INAV only sends changed data. The change that @Jetrell mentioned sends a full frame at specified intervals.

However, when you switch OSD layouts, that forces a completely redraw. So the OSD locking up sounds like more of a DJI issue, as the data will be getting redrawn from scratch by INAV.

MrD-RC, 2023-02-07:
The default value should be fine.

closed 2023-03-06 by DzikuVx