iNavFlight/inav #8766: Text corruption on OSD artificial horizon.

MonkZy73, 2023-02-03:
The artificial horizon line on my OSD has flickering, corrupted characters. All other OSD items are displaying perfectly. I thought it may be electrical noise so have shortened and re-twisted any signal wires. I have yet to maiden this 5" quadcopter.

[VIDEO](https://www.youtube.com/watch?v=M6dSpr_4YYM)

My FC is a SpeedyBee F405 v3 (OSD is AT7456E). Running iNav 5.1.0

version : INAV/SPEEDYBEEF405V3 5.1.0 Sep 13 2022 / 13:07:39 (af75e526)

[CLI 'dump'](https://pastebin.com/jLpuBPzs)

OptimusTi, 2023-02-03:
Try reloading the font file.

MonkZy73, 2023-02-03:
I have tried loading various font files but they all do the same thing. Odd letters appearing instead of the dots/lines. I have scoured the net, it seems no one has ever had this occur. My next test will be to flash betaFlight and see if I get the same glitches on the artificial horizon. I really wanted to use iNav on this quad though :(

Jetrell, 2023-02-03:
@MonkZy73 Are you sure you have the same version of the Configurator and firmware. Both in your case have to be 5.1.
When you say you tried various fonts.. Do you also mean you uploaded the font package for 5.1 first?

[image]

MonkZy73, 2023-02-04:
I was using the 5.1 configurator, with 5.1 firmware. Initially I flashed the firmware using configurator 6, and quickly found it was the wrong configurator. I have since re-flashed the drone with 5.1 using 5.1 configurator. Re-uploaded a font (with my 4S connected) and still the same corruption on the horizon line.

I have also tested the FC with BetaFlight, the artificial horizon is perfect on BF. This rules out the OSD chip or FC being faulty, I think.

I will have a play with BF, but I will likely re-flash iNav later today. (i want the logic/programming features and waypoints). So far it is only the horizon line I have trouble with.

MonkZy73, 2023-02-04:
I have done some further investigation/testing on this issue.

I configured betaFlight up to the point I would be able to arm and fly the vehicle. The artificial horizon works fine.

[betaFlight OSD (cap on)](https://www.youtube.com/watch?v=EzRBbmAa0UM)
[betaFlight OSD (cap off)](https://www.youtube.com/watch?v=nNXzLtnE9Yw)

I took a backup of the settings and a CLI dump [(pastebin)](https://pastebin.com/iTBmd4Tx).

One thing of note was the betaFlight setting **set max7456_clock** on the F405 v3 target this is set to 'NOMINAL'. This setting reduces the SPI clock speed to the OSD to assist some max7456 clones. My FC is fitted with a clone. There is a [thread](https://github.com/betaflight/betaflight/pull/4150) on the betaFlight github discussing the inclusion of the setting.

I have re-flashed with iNav 5.1.0, re-uploaded the default font and the corruption has returned.

[iNav 5.1.0 OSD (cap on)](https://www.youtube.com/watch?v=rgPRbeM0GIM)

I am going to upgrade to iNav 6.0 next and set up for a maiden. I have turned off the artificial horizon in my simulator. I am doing more investigation into the OSD spi bus speed setting on iNav.

Pedrolomel, 2023-02-04:
I confirm. I have similar issues with OSD displaying. See link below:
https://github.com/iNavFlight/inav/issues/8733

MonkZy73, 2023-02-05:
My conclusion is that the SB F405 v3 has been designed for cost effectiveness. They have paired the SD card with the OSD on the SPI bus. This allows for a reduced speed bus to not affect the gyrometer (which is usually paired with the OSD). In betaflight you can adjust the OSD bus speed (usually to the detriment of the gyro readings). I think SpeedyBee made a cool cost cut to the design. The cleanflight code had overclocked the OSD chips in order to up the bus rate on the gyro, rendering standard 7456 chips useless. Standard 7456 chips work just fine when clocked correctly.

SpeedBee is not doing their customers a service by claiming compatibility to iNav.

This bug is probably a very complicated feature request. Add a setting to make ancient OSD chips work with iNav.

OptimusTi, 2023-02-05:
Could you please check if INAV 4.0.0 has the same issue?

MonkZy73, 2023-02-05:
I am more than happy to flash iNav 4.0 for a test, unfortunately there is no target for the speedybee F405 v3. How do I proceed? Will I need to compile the .hex myself?

MonkZy73, 2023-02-05:
I am in a position to compile  iNav. I can only find the target.h file for iNav 6.0. I took a look at maybe preparing a target for iNav 4 but it is beyond my knowledge and ability, I am afraid. If someone could make a 4.0 target from the details in the 6.0 target and pastebin the code, I can compile and test.

konikav, 2023-02-22:
Looks like i have the same problem with artificial horizon in INAV6, any ideas how to fix it?

closed 2023-10-27 by DzikuVx