iNavFlight/inav #8773: HAKRC F7230D with HAKRCK DF722 firmware does not recognize accelerometer and gyro.

PaxHitech88, 2023-02-06:
## Current Behavior
After loading the HAKRCKD722 firmware accelerometer and gyroscope are not recognized in INAV 6.0. 
## Steps to Reproduce

1. Flash Firmware in HAKRC F7230D
2. Calibrate

## Expected behavior
Correct recognition of the accelerometer and gyroscope of the FC HAKRC F7230D in order to configure the FC correctly 

## Suggested solution(s)
Implement the FC HAKRC F7230D or have it recognize the accelerometer and barometer.

## Additional context
I request this for personal use after having asked and searched in various INAV groups on the social Facebook, having asked in telegram and discord. After making these requests I noticed that many users have the same HR and would like to use INAV on their quads too.

I attach here the link to the site where I bought the FC in order to see the FC. 
https://it.aliexpress.com/item/1005004859612345.html?spm=a2g0o.order_list.order_list_main.52.32c83696bTWzlT&gatewayAdapt=glo2ita
I would add that I have tried several times to load the compatible firmware (see MATEK F722, MATEK F722SE, MATEK F722mini, HAKRCKD722, etc.) but to no avail, currently I have loaded the Bataflight 4.4 HAKRC F722 firmware with accelerometer and gyroscope set manually (ACC_SPI_MPU6500 GYRO_SPI_MPU6500) at the time of the firmware flash .

- FC Board name and vendor: HAKRC F7230D
- INAV version string: 6.0.0

PaxHitech88, 2023-02-08:
I managed to solve the problem, now the HAKRC F7230D board is recognized correctly, the accelerometer and the gyroscope are recognized and calibrated correctly. I did a flight test and it flies fine. I have yet to test the RTH and failsafe.

PaxHitech88, 2023-02-08:
To make INAV work correctly with this card you need to load the firmware for MATEK F722 SE, then set the accelerometer and gyroscope manually from the CLI with the commands:
set gyro_to_use=1
set acc_hardware=MPU6500
save and restart.
Then just continue with the configuration in the traditional way.

closed 2023-02-08 by PaxHitech88

htagv, 2023-02-13:
[image]
没有找到黑匣子芯片

PaxHitech88, 2023-02-13:
I'm having the same problem, I'm trying to fix it as soon as I can update.

PaxHitech88, 2023-02-14:
> 
> [image]
> 没有找到黑匣子芯片

I solved the problem by uploading the firmware for matek F722 mini, the settings I wrote above are still valid.

htagv, 2023-02-14:
你真棒

htagv, 2023-02-18:
Unable to unlock，After unlocking, the drop rings, but there is no error in the log
[image]

PaxHitech88, 2023-02-18:
> Unable to unlock，After unlocking, the drop rings, but there is no error in the log
> [image]
> 

Is the card firmware correct?  did you set accelerometer and gyroscope correctly?  did you set the barometer?  can you send the configuration to see how they are set up?

htagv, 2023-02-18:
Set it up, it turns out that the throttle value on the remote control is high, the default is 1100, and it can be unlocked by changing it to 1000

htagv, 2023-02-18:
How do I set throttle travel inside INAV?

htagv, 2023-03-13:
如何设置低电压保护，自动返航

whwxxstsyd, 2023-06-15:
> 如何设置低电压保护，自动返航

兄弟，你这板子的陀螺仪是6500的吗

PaxHitech88, 2023-06-15:
no, it's a MPU 6000

whwxxstsyd, 2023-06-15:
> no, it's a MPU 6000

No wonder，Thanks