iNavFlight/inav #8778: BZGNSS GPS modules only running at max 5Hz

Pairan, 2023-02-08:
Currently the new M10 GPS modules from **BZGNSS** (e.g. `BZ-121`, `BZ-181`, `BZ-251`) only run at 5Hz although they are capable of 10Hz refresh rate (when not ALL systems activated if I read right). The GPS units are defined as this:

- Chip:u-blox M10050(Ten Generation Chip)
- Pattern: GPS,GLONASS,BDS,GALILEO,SBAS,QZSS(Default GPS+GLONASS+BDS joint positioning)
- FLASH:Supported
- Frequency:GPS L1,GLONASS L1,BDS L1,GALILEO L1,SBAS L1,QZSS L1
- Power:5V
- Size:12x16x4.5mm
- Interface:PCB Pad
- Weight:2.2g
- Antenna:Ultrathin Ceramic Antenna (oil spraying process, prevent scratching and oxidation)
- Receiving channels:72 search channels
- Baud rate:115200dps
- Output Protocol:NMEA / UBLOX (Dual protocol)
- Output Frequency:1Hz-10Hz. Default 10Hz
- Speed accuracy:0.05m/s
- Horizontal Positioning Accuracy: 2D ACC 1.5m(open air)
- Receiving Sensitivity:Trace-162dBm / Capture -160dBm

INAV works fine with these units just cant get up at 10Hz by now as it seams ...

[image]

## Current Behavior
INAV (5.1) operates these GPS units at max 5Hz

## Steps to Reproduce
1. connect the GPS unit accordingly
2. set UBLOX7 protocol
3. take a look at the refresh rate

## Expected behavior
I would like to get it up at 10Hz like on the older modules (Beitian BN-180). After having set the rates to 10Hz in U-Center (2) and saving it to flash, it gets reset after INAV has booted up. ... am I doing anything wrong here?
[image]
_After INAV has booted up, the refresh rate is back at 5Hz / 200ms_

## Suggested solution(s)
Set the GPS refresh rate up to 10Hz instead of 5Hz

## Additional context
Perhaps the setup for the GPS units have changed with M10 chips

#### INAV/MATEKF405SE 5.1.0 Aug 19 2022 / 12:05:31 (76f22b25)
#### GCC-10.2.1 20201103 (release)
[image]

stronnag, 2023-02-08:
The setting for M10 is by design. Not being limited to 16 satellites is considered more important than 10Hz.

stronnag, 2023-02-08:
See #6269

closed 2023-02-08 by stronnag

Pairan, 2023-02-08:
Thanks for the update, mate ... wasn't aware of it!