iNavFlight/inav #8781: INAV 6 Release Candidate 1: Speedybee F7V2: Port settings not saved

fpvbam, 2023-02-09:
## Current Behavior
Settings in the ports tab are lost eveytime after saving.

## Steps to Reproduce
Visit ports tab, change something, save.

## Expected behavior
Saved Settings should be displayed.

## Suggested solution(s)
?

## Additional context
After full flash erase, I tried version 5.1 and works as expected.

---

- FC Board name and vendor:
- INAV version string: - SPEEDYBEEF7V2

DzikuVx, 2023-02-09:
You are probably trying to save not allowed configuration. Like 2 functions assigned to a single port

fpvbam, 2023-02-09:
I cannot reproduce it. Btw, I had problems to flash INAV 6 on Mac (Big Sur) with Configurator 6, no DFU provided. Used STMCubeProg.

closed 2023-02-09 by fpvbam