iNavFlight/inav #8783: FW Climb First OFF, turn first, then gain altitude

tiriad, 2023-02-09:
## Current Behavior
When in fixed wing "Climb first" is OFF, if the altitude is lower than the set altitude, the engine will accelerate a lot and it is also difficult to turn while increasing in altitude.
## Desired Behavior
In fixed wing when RTH climb first OFF,  turn home first and then start to recover altitude

## Who does this impact?
Fixed-wing users

## Additional context
The fixed wing will not perform excessive acceleration and the possibility of loss of control will be avoided.

Jetrell, 2023-02-10:
@tiriad I'm not knocking your request.
But if your plane is having trouble turning and climbing at the same time, then it maybe beneficial to optimism the below setting to suit your planes thrust to weight ratio... Producing too little thrust, to suit the task and conditions, is likely why its having troubles.

`nav_fw_max_thr`
`nav_fw_pitch2thr`
`nav_fw_climb_angle`

Jetrell, 2023-02-10:
@tiriad I'm not knocking your request.
But if your plane is having trouble turning and climbing at the same time, then it maybe beneficial to optimize the below setting to suit your planes thrust to weight ratio... Producing too little thrust or too steep of climb angle, to suit the task and conditions, is likely why its having troubles.

`nav_fw_max_thr`
`nav_fw_pitch2thr`
`nav_fw_climb_angle`

tiriad, 2023-02-10:
In my particular case I have a high thrust to weight with a 7 inch propeller and a high Kv motor and it tries to turn too fast. I have already adjusted down  `nav_fw_max_thr` , `nav_fw_cruise_speed` although maybe not enough. Since `nav_fw_pitch2thr_smoothing` was introduced I have left the default `nav_fw_pitch2thr` setting but I may have to lower it by half again. 
In any case, I think it would be smoother to do a turn and  head home before starting to recover the altitude defined in `nav_rth_altitude.`

MrD-RC, 2023-02-11:
It it is turning too fast, you should look at the maximum navigation angles. Reducing the maximum bank angle will make for slower turns. Also, you may want to check your loiter radius isn't too tight.

tiriad, 2023-02-11:
The only issue is with the throttle, not with the banking `nav_fw_bank_angle` which is already reduced from the default setting. I am going to reduce the `nav_fw_pitch2thr` at less than half and I think it will mitigate my problems. I have understood the operation and the maximum throttle change I want in the climb angle (`nav_fw_climb_angle` + `fw_level_pitch_trim` - `nav_fw_pitch2thr_threshold`) * `nav_fw_pitch2thr`
Anyway if any developer is interested in implementing this proposed option I will be grateful and it won't be necessary to do so much tuning.

tiriad, 2023-02-11:
I am not sure if this is the case but since the introduction of the `fw_level_pitch_trim` in versión 3 instead to use `align_board_pitch` as previously, `nav_fw_pitch2thr`  , `nav_fw_pitch2thr_smoothing` and  `nav_fw_pitch2thr_threshold` could continued using for calculations the absolute `Pitch` and not the corrected `Pitch` with `fw_level_pitch_trim` so it may affect the correct working  of the functions and therefore could produce more acceleration than before when ascending.

closed 2023-02-19 by tiriad