iNavFlight/inav #8785: Motor without dshot protocol input

0AleBaldi0, 2023-02-09:
FC f405 matek hdte.
ESC f55a proII t-motor.

Recently I've replaced my FC (T-motor f70) cause damaged it, with the one above . Reweld everything and go to Inav for programming. Activated output motors and choose "dshot 600" like the "T-motor f70" . When I've tested them, one motor didn't move.
Plug the battery on the quad and the starting sound repeat itself many times with all beep of control, but like for all motors. So I've checked motors with blheli 32 and one motor haven't dshoot protocol input( the one who didn't move before).
So maybe this is the problem but don't now how to fix. When choose multishot/oneshot the problem is fixed but I want dshot for turtle mode. Don't know why on the first FC this problem there wasn't . Thanks for help

0AleBaldi0, 2023-02-09:
Video of the starting sound with dshot 600

https://drive.google.com/file/d/1i_nH6X22hEqWqObbZg75YUIX0ftcmvYA/view?usp=drivesdk

mmosca, 2023-02-10:
Did dshot 600 work with the previous esc?

Is the esc new?

It could be  an esc issue. Try reflashing blheli32 to the esc.

0AleBaldi0, 2023-02-10:
The esc is the same. With the previous FC was already using dshot 600.
I've updated the ESC to the last version of blheli_32 (32.9) but the problem remains

0AleBaldi0, 2023-02-10:
What I see when check with blheli32

https://drive.google.com/file/d/1jIJvhcIgo96nhtuVgNeQJgiPCyxuwZxe/view?usp=drivesdk

0AleBaldi0, 2023-02-10:
Some weeks ago I've burn a motor ( the one who doesn't work now)and changed with a new one and with the FC f70 T-motor it was working good. But not with matek f405

0AleBaldi0, 2023-02-10:
> Did dshot 600 work with the previous esc?
> 
> Is the esc new?
> 
> It could be an esc issue. Try reflashing blheli32 to the esc.

Forgot to reply    ^^^

tiriad, 2023-02-10:
It might be related to this open issue #8089

0AleBaldi0, 2023-02-10:
> It might be related to this open issue #8089

Thanks. Now I'll try to rewire something let you know

mmosca, 2023-02-10:
@0AleBaldi0 you can try my pull request: #8788 

You won't need to rewire the FC but as a side effect S9 will also be a motor output.

0AleBaldi0, 2023-02-10:
I've to download the configurator ?

0AleBaldi0, 2023-02-10:
And it's done ?

mmosca, 2023-02-10:
> And it's done ? Cause I've already cut motor S3 S4 for slide them respectively in S4 S5 .......

You need to use the 6.0 RC2 configurator.

You can also try the solution mentioned in 8089.

0AleBaldi0, 2023-02-10:
> You need to use the 6.0 RC2 configurator.

Ok later I'll try thx

0AleBaldi0, 2023-02-10:
> You need to use the 6.0 RC2 configurator.
> 
> You can also try the solution mentioned in 8089.

you've resolve me 2 problems that weren't in list but thanks XD.
1. starting sound fixed. Does not repeat control sound. One clear and  with all its beeps.
2. Led issue fixed. Problem: when try to ordering wire it says that i've 0 numbers remains. 

Unfortunately when test motors with dshot600 the same motor didn't work.

You know any command to say at the fc that it has to use S5 instead of S3 ? cause when i slided cables S3 and S4, without using "S3 PAD" i could run all motors

0AleBaldi0, 2023-02-10:
> You need to use the 6.0 RC2 configurator.

i haven't do nothing apart installing configurator and flashing fc. But didnt understand  when you talking about "side effect s9 "ecc..

mmosca, 2023-02-10:
Did you test with the firmware from the assets in the pull request, or did you just pick the firmware from the configurator list?

If you did not use the hex from the pull request, try to load this one:

[inav_6.0.0_MATEKF405TE_S3DSHOT_ci-20230210-673d743.zip](https://github.com/iNavFlight/inav/files/10711695/inav_6.0.0_MATEKF405TE_S3DSHOT_ci-20230210-673d743.zip)

mmosca, 2023-02-10:
> > You need to use the 6.0 RC2 configurator.
> 
> i haven't do nothing apart installing configurator and flashing fc. But didnt understand when you talking about "side effect s9 "ecc..

To try and make S3 work with DSHOT, I had to change the timer used, with the change S9 and S3 share a timer, so they need to have the same function, so S9 ends up being a motor output as well.

0AleBaldi0, 2023-02-10:
> Did you test with the firmware from the assets in the pull request, or did you just pick the firmware from the configurator list?
> 
I've tested it with the hex file found in your suggestion 8788 ( the one with S3dshot in the title) now try your zip

0AleBaldi0, 2023-02-10:
> > Did you test with the firmware from the assets in the pull request, or did you just pick the firmware from the configurator list?
> 
> I've tested it with the hex file found in your suggestion 8788 ( the one with S3dshot in the title) now try your zip.

Update. Yes I've used the one that u mentioned

0AleBaldi0, 2023-02-10:
> To try and make S3 work with DSHOT, I had to change the timer used, with the change S9 and S3 share a timer, so they need to have the same function, so S9 ends up being a motor output as well.

How can I change timer ? It is "motor timing" in blheli ?

rmaia3d, 2023-02-11:
If you got Matek webpage for this FC, there's a note saying that S3 won't work with DSHOT due to DMA clash (internal MCU resource conflict):

[image]

Same is true for the F-405TE:

[image]

@mmosca solution to change the timer used for the S3 output can indeed solve this issue. I haven't looked at his code in detail, but it makes perfect sense.

rmaia3d, 2023-02-11:
> > To try and make S3 work with DSHOT, I had to change the timer used, with the change S9 and S3 share a timer, so they need to have the same function, so S9 ends up being a motor output as well.
> 
> How can I change timer ? It is "motor timing" in blheli ?

No, it has nothing to do with motor timing or BLHeli. The timer he mentioned is an internal MCU (STM32F405) resource, it's basically what allows the MCU to output precisely timed signals, like DSHOT requires. There's a limited number of them available at the MCU, and several features end up sharing a same timer. If two features don't use the same timer concurrently, then there's not much of a problem. But if two features try to use the same timer at the same time, and require different settings for said timer, that's when the conflict/clash occurs, and one of the features will simply not work. The solution is to use separate timers for each feature, if available.

0AleBaldi0, 2023-02-11:
And how can I use separate timers for each feature? There's a cli code ? Sorry but with programming I'm bad... In 8788 found something but don't now how to proceed.

0AleBaldi0, 2023-02-11:
or maybe would be good also solution in 8089. This fc will remains forever with inav, no change. But the problem is that i dont know how to change used motors with others S output (like S6). 

now the wires are in S1 S2 S3 S4, so if its easy ( relatively ) i could try your suggestions @rmaia3d @mmosca. if its not, i dont use S3 S5 S7 and peace. i have only to make it fly with dshot and its all ok

mmosca, 2023-02-11:
If you tried my build and it still didn't work, either switching to oneshot or using the solution described in 8089 is probably better.

0AleBaldi0, 2023-02-11:
Ok, you know how can i change default motors with another S output, like for the motor 3 use S4 and for motor 4 use S5 ( when i added a new mixer rule for S5 i could run the motor in S5 with dshot and no problem) so this DMA clash dont make problem with S5 in my case no ?

closed 2023-02-14 by DzikuVx