iNavFlight/inav #8786: iflight blitz 1.6 pwr selection not working, irc tramp

Dteksolutions, 2023-02-10:
## Current Behavior
iflight blitz 1.6 vtx, doesnt change pwr lvls after 1&2. it may say that its on lvl 4 but the indicator lights only say lvl 2.

## Steps to Reproduce

1.install the above vtx
2.select the icr tramp on the port that its on
3. change pwr lvl in configurator. 1,2,3,4,5
4. Watch leds on the vtx.

## Expected behavior
I expected the pwr lvl to increase in conjunction to the lvl selected. 

## Suggested solution(s)
irc tramp tables? Let us load our own. Anything that lets the vtx function on the lvl its supposed to be on.

## Additional context

https://pastebin.com/cBxhmLGp

---

- FC Board name and vendor:
-  INAV/MATEKH743 5.1.0 Aug 19 2022 / 12:34:02 (76f22b25)
# 
- INAV version string:
- GCC-10.2.1 20201103

nkorls, 2023-04-04:
same problem
HGLRC Zeus nano VTX

fermiums22, 2023-04-04:
I have already raised this issue. While you have to edit the source itself =(
https://github.com/iNavFlight/inav/issues/8340#issuecomment-1465248387

b14ckyy, 2024-03-21:
Duplicate

closed 2024-03-21 by b14ckyy